kth-ros-pkg/yumi #7: Some files are missing

sinamr66, 2017-11-21:
Hi All,

First of all, let me thank you all for creating this great package and the wiki!... I was following the wiki to set-up the framework, however, I failed to find the following files in the package:

EGM/TRobEGM.mod
EGM/TRobMain.mod
EGM/TRobSynchronization.sys
EGM/TRobSG.sys

Can you please help me?
Moreover, while defining a new task in RobotStudio, its type can be either normal or Semistatic/static. However, if one chooses Static, there is no way to define the trust level. Hence, I was wondering how the type of the tasks and defined Normal while Trust Level is specified in the following instruction:

https://github.com/kth-ros-pkg/yumi/wiki/Firmware-setup-(article-in-development)

Thanks in advance for our help.
Best
Sina

sinamr66, 2017-12-11:
Hi Guys,

No News?
I've also checked all the backup files, they don't exist.... Are they located in a private repo? OR sth else?
ANY COMMENT IS HIGHLY APPRECIATED :+1: 

Best,
Sina

diogoalmeida, 2018-01-12:
Hi Sina, 

Thank you for your comments, and sorry for the late reply. I am afraid that EGM is a proprietary interface for using ROS with Yumi, so we cannot provide it for free. I spoke yesterday to our research engineer @YoshuaNava, which mentioned he might make an option to use other interfaces. I'll try to make sure that future comments are addressed faster.

Let me know if this is still relevant to you.

Best,
Diogo

sinamr66, 2018-01-12:
Hi Diogo,

Thanks a lot for your reply. It was really helpful. If you don't mind, may I ask how you guys purchased it? Can you please provide more details regarding the name and the version of the software/interface?

Thanks a lot for all your help.

Best,
Sina

diogoalmeida, 2018-01-12:
As a lowly PhD student, the purchasing details are a bit above my paygrade :). @YoshuaNava Should know more about the software/interface details, so I will ask him to write a nice reply.

YoshuaNava, 2018-01-12:
Hello Sina,
Sorry for the delay in replying.

I'm preparing some changes that will allow you to build this package without EGM/RWS. RWS is a library for communication with the robot internal controller, that allows you to do position control, get images from the grippers cameras, get the state of the robot, etc. EGM is a library for direct velocity control that runs at about 250Hz. To launch EGM it is mandatory to have RWS.

Regarding the distribution of these libraries (and their RAPID modules), we have asked ABB, but they don't allow us to put the scripts in the repo. Even more, for running RWS and EGM they need to flash the YuMi firmware.

Regarding the acquisition of RWS/EGM, I suggest that you contact ABB Sweden, and if possible, the EGM developer Jon Tjerngren @jontje

Hope this helps. Please let me know if you have any other questions.

Yoshua

lz89, 2018-01-12:
Hello @YoshuaNava and @sinamr66 ,

We (Imperial College) recently purchased the the EGM module from ABB UK. The local team was not aware of the "EGM for YuMi" as currently it is not officially supported. I contacted the ABB engineer Jon Tjerngren @jontje who explained to me that only EGM joint control is supported for YuMi.

@YoshuaNava ABB UK support team sent me a file (named: 14000-500942_RobotWare.rlf), and asked me: "build a system in RobotStudio using installation manager then add this file to the license tab". Do you have any experience on this? If I manage to build the system with EGM myself successfully, can I get the related RAPID code you are using?

Thanks,
Lin

YoshuaNava, 2018-01-14:
Hey,
@lz89 Great to hear that you could acquire EGM. Indeed, EGM is a product in development at this moment. What I've heard from them is that they need to flash the internal controller, so I'm not 100% sure.

One of the most important things to check once you have installed the EGM license is the robot network config. You need to make sure that the IP for receiving messages is set, under UDP protocol. We normally use ```192.168.125.50``` in our YuMis. I can tell you more about this tomorrow, when I'm at the lab.

@sinamr66 I just finished setting up the cmake files so that you don't need EGM/RWS for building this package. It limits quite a lot the amount of things that you can do (velocity control and gripper cameras are not accesible without RWS and EGM), but you can still use the ABB driver on a real YuMi, and Gazebo.
I'll be testing this latest changes, and notify you when they are good to go.

Hope this helps. Please let me know if you have other questions.

Yoshua

jontje, 2018-01-14:
Hello everyone,

Here are some answers/corrections that I hope clarifies a few things. And sorry for the “bulky” reply.

**----- Questions -----**
@sinamr66
Maybe you already received all you needed concerning these, but I added them anyway:
1.	TRobEGM.mod, TRobMain.mod, etc.:
They are currently not open-source. See “Concerning potential open-source” below for a bit more info.
2.	If you have a Static task:
There shouldn’t be a problem to set the trust level. I.e. in Configuration -> Controller -> Tasks.
3.	EGM_JOINT_START, EGM_POSE_START, etc.:
They are IO-signals, and not tasks.

@lz89
The .rlf file is a license file.

You use it when you install a new RobotWare system (on a real robot controller). E.g. in RobotStudio you can do something similar to this:
1. Open Installation Manager (under the Controller tab), and choose your robot controller.
2. If needed: Free space by deleting an old system.
3. Add a new system:
3.1. Under Products:
3.1.1. Press Add... -> Choose the RobotWare version.
3.1.2. (Optional) Press Add... again -> Add SmartGripper product (if you have the grippers).
3.2 Under Licenses -> Press Add... -> Find and choose the .rlf file.
3.3. Under Options -> This should already be filled out.
3.4. Under Confirmation -> Press Apply.
4. Wait until the intallation finishes.

**----- Corrections/clarifications -----**
**Concerning RWS and EGM:**
- Robot Web Services:
Is a free feature of ABB robot controllers (requires RobotWare >= 6.0). Can be used for general interaction with ABB robot controllers (e.g. setting/reading IO-signals, uploading files, updating/reading configurations, starting/stopping RAPID execution, writing/reading RAPID variables, etc.). See the [documentation](http://developercenter.robotstudio.com/webservice/api_reference) for more info.
- Externally Guided Motion:
Is a released product, and it requires an optional license, but it is only officially available for 6-axis robots. Used for direct motion control of the robot (position and/or velocity commands). There exists a prototype extension (for RobotWare >= 6.03), with limited EGM functionality for YuMi robots. It is the business unit that decides if it can be provided or not. I.e. contact the local ABB Robotics sales organization.

Note: RWS and EGM are completely decoupled, so it is perfectly fine to use one without the other. Both are also platform independent, so they can be used with e.g. Ubuntu or Windows etc.

**Concerning potential open-source:**
The RWS and EGM products specifies the interfaces (i.e. used communication channels and protocols) exposed by the robot controller. They do not provide anything for e.g. the external side, see the sketch below for a rough overview:

[image]

- RWS and EGM libraries:
I have implemented libraries for the external side, i.e. the RWS and EGM C++ libraries. They help with setting up clients respective servers, parsing messages as well as providing some user friendly APIs. And as above, they are completely decoupled.
- RAPID code:
I have also implemented a basic RAPID program (i.e. consisting of the files: TRobEGM.mod, TRobMain, etc.). This program handles e.g. the EGM RAPID instructions. It is not mandatory to use this program, but it can simplify a few things.
- Concerning when:
An internal open-source contribution process is currently ongoing, and it is in its final stages. But the exact time for making it available is not set, but hopefully soon.

I hope this helps.

Best regards,
Jon

diogoalmeida, 2018-01-14:
Thanks for the reply @jontje! Really informative.

YoshuaNava, 2018-01-14:
Hey,
@jontje thank you very much for your reply!

I might be mistaken, but with respect to the EGM and RWS libraries, they may be used standalone without a problem. But, without RWS you might not be able to set parameters  that are critical for EGM performance (as we do [in here](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_hw/src/yumi_hw_egm.cpp#L303)).

jontje, 2018-01-15:
@YoshuaNava: Ah, yes, in your case it is a good fit for RWS. I was referring to a more general case, and I should have mentioned that. My bad :)

There are other ways to do something similar, e.g. with RAPID sockets or the IO-board and interrupts. However, I practically always prefer to use RWS.

sinamr66, 2018-01-15:
Dear All,

I really appreciate all of our help. the replies help me a lot to understand the control architectures. 

Thanks  a lot..:)

Best,
Sina

lz89, 2018-01-15:
@YoshuaNava Based on @jontje clarification, is it mean we can't use the EGM feature of this repo at the moment? Before the EGM RAPID code become public, the options are: 1) Orebo repo with ABB driver (yumi_support) or RWS (yumi_hw); 2) ABB driver of this (kth) repo?

Thanks!

YoshuaNava, 2018-01-17:
@lz89 Exactly. As Orebro University is not going to continue developing the YuMi package, we are setting ours at KTH to provide the same features without the compilation problems that occur when you don't have RWS/EGM.

I made some changes during the weekend in that direction. I'll be testing them in all our branches this week. Nonetheless, if you need to introduce them immediately, check the changes made in [optodaq_sensors/yumi_hw CMakeLists.txt](https://github.com/kth-ros-pkg/yumi/blob/optodaq_sensors/yumi_hw/CMakeLists.txt)

YoshuaNava, 2018-01-17:
Ok. I've just tested building the package without EGM/RWS, and it works fine. Check issue #11 for more details.

closed 2018-01-17 by YoshuaNava

lz89, 2018-01-17:
@YoshuaNava Great, thanks a lot! Just to confirm:
I should use the `egm_modifications` branch of this repo and use RAPID in the yumi_hw to setup the RWS control interface for YuMi without the EGM feature.
I will test this and let your know if there is any issue.

galou, 2018-07-31:
I'm interested in getting the images from the gripper's camera. I don't need EGM. Is there a chance to get the `abb_rws_interface` package from somewhere?

Do you know any alternative to get the images from the camera in ROS (I shortly tried https://github.com/HumaRobotics/modbus without success yet).

Thanks for any info,
Gaël

jontje, 2018-08-07:
@galou: I haven't used the SmartGripper camera for quite some time, but there are some options.

One option is to have a job on the camera, which stores the image in the robot controller's memory. Then the RWS **file service** can be used to retrieve the image. An issue with this is that it isn't very fast.

Another option might be to investigate if the RWS **Integrated Vision (IV) service** have any new features that allows for direct access. I haven't had time to look into it myself.

If I remember correctly, then @YoshuaNava have done some work for accessing camera images, which might be using some other approach.

Finally, you can use [https://github.com/ros-industrial/abb_librws](url) instead of `abb_rws_interface`. `abb_librws` is the successor of  `abb_rws_interface`, and it is open-source.

I hope this helps!