kth-ros-pkg/yumi #10: MoveIt! Reaches joint limits when comanding yumi.

diogoalmeida, 2018-01-12:
It is happening on the ridgeback_yumi, but I am assuming the true issue is in this repo. When using MoveIt!, some of the trajectories will cause Yumi to reach joint limits. This should **never** happen.

hkaraoguz, 2018-01-12:
The solution I had was to modify the urdf file under yumi_hw and decrease the joint limits

diogoalmeida, 2018-01-12:
Can you point me to the values you used?

hkaraoguz, 2018-01-15:
https://github.com/kth-ros-pkg/yumi/blob/project/FACT/yumi_description/urdf/yumi.urdf

YoshuaNava, 2018-01-17:
Hey,
Right now the URDF containts the joint position and velocity limits specified in the [YuMi datasheet](https://github.com/kth-ros-pkg/rplfloor7_manuals/raw/master/ABB%20YuMi/ROBO0317_Yumi_print.pdf).

I had also thought about reducing the joint limits in the URDF. However, I didn't do it because I thought that for correctness we should follow ABB's specification.

Nonetheless, it's a solution that has already been put into practice by @hkaraoguz and another team in the lab. What is your opinion on this? Should we modify the values or keep them as they are?

gavanderhoorn, 2018-01-17:
@YoshuaNava: this could be a degree to radian conversion thing.

In my experience, converting to radians and then flooring to 4 digits typically works.

diogoalmeida, 2018-01-17:
It feels like @gavanderhoorn solution is worth a shot @YoshuaNava :)