kth-ros-pkg/yumi #14: Undefined substitution argument yumi_setup

lz89, 2018-01-18:
I have installed the latest egm_modifications branch and was able to compile the code without any error.
Note that I don't have EGM yet, so I think the CMake has disable the related features automatically.
I believe I am using the RWS (without EGM) interface.
I tried to start the connection from:
```
roslaunch yumi_launch yumi_traj_pos_control.launch
```
however, I got an error:

> None Undefined substitution argument yumi_setup 
> when evaluating expression 'yumi_setup == 'default''
> XacroException(None,)
> when processing file: /home/xxx/dev/catkin_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro
> Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/superbeast/dev/catkin_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro' arms_interface:=PositionJointInterface] returned with code [2]. 
> 
> Param xml is <param command="$(find xacro)/xacro.py '$(find yumi_description)/urdf/yumi.urdf.xacro' arms_interfa

I also tried to run (based on the wiki):
```
cd $YUMI_WS/src/yumi/yumi_description/urdf/
rosrun xacro xacro yumi.urdf.xacro arms_interface:=VelocityJointInterface grippers_interface:=EffortJoinInterface yumi_setup:=robot_centric -o yumi.urdf
```

Please let me know what I should do in order to establish the connection with YuMi using the RWS interface.

YoshuaNava, 2018-01-19:
Hi @lz89,
First of all, thank you for your feedback.

**yumi_setup**
yumi_setup is a parameter that we use to build a yumi URDF without any root links. It isseful in particular for people in our lab that want to build the URDF without looking at the robot macros and additional accessories that we set in the xacro files.

In order to solve that problem, add [this line](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_launch/launch/yumi_traj_vel_control.launch#L18) and [this other line](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_launch/launch/yumi_traj_vel_control.launch#L19) to the ```yumi_traj_pos_control.launch``` file. 

(I will also fix the absent definition of yumi_setup in the launch files as soon as I get to the lab)

**RWS**
To run RWS you should load [RWS](https://github.com/kth-ros-pkg/yumi/wiki/Firmware#robotware-services-rws---position-control) and [grippers](https://github.com/kth-ros-pkg/yumi/wiki/Firmware#grippers-and-cameras) RAPID modules into your YuMi, using RobotStudio.
Then, you should [start the RAPID modules](https://github.com/kth-ros-pkg/yumi/wiki/Live-robot#starting-the-rapid-scripts-important) using the FlexPendant. You will hear the grippers opening and closing. That's a calibration routine.
Finally, you should run the command you mentioned before in a terminal:

```
roslaunch yumi_launch yumi_traj_pos_control.launch
```

If everything goes well, you should see YuMi loaded in rviz, and the robot model displayed in the same position as your physical YuMi.

Please let me know if this helps.

Yoshua

lz89, 2018-01-19:
Hi @YoshuaNava , thanks a lot for the reply.
I modified `yumi_traj_pos_control.launch` line 18,19 to:
```
<!-- URDF / SRDF parameters -->
<arg name="yumi_setup" default="default"/>
<param name="robot_description" command="$(find xacro)/xacro.py '$(find yumi_description)/urdf/yumi.urdf.xacro' arms_interface:=$(arg arms_hardware_interface) grippers_interface:=$(arg grippers_hardware_interface) yumi_setup:=$(arg yumi_setup)" />
```

The rviz can be launched but I get errors:
process[yumi/arm_controller_spawner-1]: started with pid [665]
process[robot_state_publisher-2]: started with pid [666]
**ERROR: cannot launch node of type [yumi_hw/yumi_hw_ifce_node]: can't locate node [yumi_hw_ifce_node] in package [yumi_hw]
ERROR: cannot launch node of type [yumi_hw/yumi_gripper_node]: can't locate node [yumi_gripper_node] in package [yumi_hw]**
process[yumi_joint_state_publisher-5]: started with pid [667]
process[yumi_robot_state_publisher-6]: started with pid [668]
process[rviz-7]: started with pid [680]
[INFO] [1516379555.227924]: Controller Spawner: Waiting for service controller_manager/load_controller

----
Would you please also clarify (maybe add to the wiki?), what's the difference:
`yumi_pos_control.launch`, `yumi_traj_pos_control.launch`, `yumi_traj_vel_control.launch` and `yumi_vel_control.launch`. And how these launch files correspond to the different control interface. For example, I think the velocity one is only available for the RWS+EGM interface?
----
In addition, if I have a customised tool for one of the arm (connected directly instead of grasped by the ABB gripper), what shall I changed? I was using `yumi_support` (ABB driver interface), and I could simply not starting the GRIPPER RAPID on the FlexPedant. Here, if I use RWS (with or w/o EGM), what shall I do? I can see that there are few arguments in both launch the URDF files. 

Thanks a lot!

YoshuaNava, 2018-01-21:
Hi @lz89,
**yumi_hw_ifce_node**
There was a typo in the CMakeLists.txt. I pushed a fix now (https://github.com/kth-ros-pkg/yumi/commit/cb56067ea661670af179c179a02dd2197442be58). You should then be able to launch yumi_hw_ifce_node without problems.

**Launch files**
- yumi_pos_control.launch: this runs over RWS and industrial core sockets. It loads individual position controllers for each joint.
- yumi_traj_pos_control.launch: this runs over RWS and industrial core sockets. It loads a ```position_controllers/JointTrajectoryController``` for each arm.
- yumi_vel_control.launch: this runs over EGM. It loads individual position controllers for each joint.
- yumi_traj_vel_control.launch: this runs over EGM. It loads a ```velocity_controllers/JointTrajectoryController``` for each arm.

I have to say that this way of organizing the launchfiles is not the best, and we're planning on improving it soon. I was considering adding group declarations for the controllers to be loaded, as @diogoalmeida recommended for ridgeback_yumi (see an example [here](https://github.com/kth-ros-pkg/ridgeback_yumi/blob/master/ridgeback_yumi_launch/launch/live_robots.launch#L15)). So, one launchfile for EGM, one for RWS over sockets, and a corresponding parameter to choose either individual controllers or JointTrajectoryControllers.

If you have any recommendation in this regard, please let us know. Your feedback has been very valuable.

**Customized tool**

You should first define your tool as a URDF macro, as we do for the grippers (see [here](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_description/urdf/Grippers/yumi_servo_gripper.xacro#L20)). Then, you should call your macro from the yumi xacro file, as we do [here](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_description/urdf/yumi.urdf.xacro#L26). If you were to replace, say, the left gripper, you would just take our the left gripper macro and put your tool's.

On the launchfiles we just add the ip of the internal controller. It's the same we use for EGM, and we use it to retrieve the gripper states and send commands.

With respect to the GRIPPER RAPID issue, you mean that you couldn't start the corresponding RAPID script for the gripper you replaced, from the FlexPendant?

/Yoshua

lz89, 2018-01-23:
Hi @YoshuaNava, I think this issue has been solved so I will close it. I will open another issue for the customised tool.
Thanks a lot.

closed 2018-01-23 by lz89