kth-ros-pkg/yumi #17: Customised tool

lz89, 2018-01-23:
In #14, some discussion has been mentioned regarding using the customised tool.

----
From @YoshuaNava :
**Customized tool**
You should first define your tool as a URDF macro, as we do for the grippers (see here). Then, you should call your macro from the yumi xacro file, as we do here. If you were to replace, say, the left gripper, you would just take our the left gripper macro and put your tool's.

On the launchfiles we just add the ip of the internal controller. It's the same we use for EGM, and we use it to retrieve the gripper states and send commands.

With respect to the GRIPPER RAPID issue, you mean that you couldn't start the corresponding RAPID script for the gripper you replaced, from the FlexPendant?

----

Before I tried to use the gripper, I want to test the robot without any gripper.
I removed some of the gripper related configuration in both launch file and urdf macro file.
I used this repo ([link](https://github.com/JJHu1993/abb-yumi/blob/master)) as a reference where a `nogripper` launch file exists. More specifically, I modified:
`yumi_launch/yumi_traj_pos_control.launch`
`yumi_description/urdf/yumi.urdf.xacro`
`yumi_description/yumi.rviz`
so that these files do not contain any gripper related field.

The launch file can be run properly without any error and I can see following on the FlexPedent screen:

> ROS_StateServer->StateServer: Waiting for connection
> ROS_StateServer->Client at 192.168.125.205 connected

Also note that on the robot side, I didn't install any RAPID for the gripper but only the three tasks (ROS_StateServer, T_ROB_L, T_ROB_R) related to the arm are installed.

I can see following topic by `rostopic list`:

> /clicked_point
> /gplanner/constraint_marker
> /gplanner/fused_pc
> /gplanner/sdf_marker
> /initialpose
> /joint_states
> /move_base_simple/goal
> /rosout
> /rosout_agg
> /tf
> /tf_static
> /yumi/egm_control_period
> /yumi/joint_states

However, the robot in rviz is in the zero position, when I do `rostopic echo /joint_states` I see all joint values are zero.
In addition which topic shall I use for sending command to the robot? Is it /yumi/joint_states

Please let me know if I did anything wrong.

YoshuaNava, 2018-01-25:
Hello Lin,
**Grippers / joint states**
Were you able to visualize correctly the joint states of the robot before taking out the grippers-related stuff?

**Topics for sending commands**
For moving the robot arms, you need to send commands to their corresponding controllers. For example:
- If you have a position_controller/PositionController called [yumi/joint_pos_controller_3_l](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_control/config/position_controllers.yaml#L15), you'll need to publish to the topic yumi/joint_pos_controller_3_l/command (see an [Example](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_test_controllers/src/test_vel_control.cpp) with velocity controllers).
- If you have a position_controller/JointTrajectoryController called [joint_traj_pos_controller_l](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_control/config/traj_controllers.yaml#L10), you'll need to publish a JointTrajectory message to the topic /yumi/joint_traj_pos_controller_l/command. See an [Example](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_test_controllers/src/test_traj_vel_control.cpp), again with velocity controllers)

With respect to moving the grippers:
- The Gazebo simulator allows you to couple ROS controllers to it, and depending on the type of interface you load for each gripper joint (position, velocity, effort), you will be able to use different types of controllers. We employ [effort controllers](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_control/config/gripper_controllers.yaml#L5), because the physical robot actually only allows you to send effort commands.
- The physical grippers show a significant delay when executing commands, and they only allow you to send a discrete effort command every once in a while. Hence, they weren't good candidates for the implementation of a RobotHW interface. We actually set an interface of topics and services for moving the grippers from our computer.
- We worked on making the same interface for controlling the grippers, using topics and services, available in simulation and when using the physical robot. See [grippers control] for more info. 

Hope this helps. Please let me know if you have any more questions.
Yoshua

gavanderhoorn, 2018-01-25:
Slightly off-topic, but @YoshuaNava: do I understand correctly that the base yumi xacros need to be edited to attach other grippers to it?

YoshuaNava, 2018-01-25:
@gavanderhoorn Yes. With respect to the yumi xacro I refer to the [urdf.xacro file](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_description/urdf/yumi.urdf.xacro#L21) from which we call the macros for:
- YuMi with its base, body and arms.
- The grippers to be used.
- Any other add-ons, like force-torque sensors.

gavanderhoorn, 2018-01-25:
To fully exploit composibility of xacro macros, might it make sense to change this to a setup where the base yumi and grippers are modelled completely independently, and specific configurations (ie: combinations of base yumi + specific grippers, or no grippers) are defined in a 'higher level' xacro?

That xacro then essentially only couples a few macros to get to a complete tree.

This would increase reusability of the xacros and also avoid the situation where consumers of your xacros need to go in and edit them - which becomes a problem if/when you release them as ROS packages and they are installed in non-world-writable locations.

lz89, 2018-01-25:
@YoshuaNava 
I try to install the RAPID code for the left gripper (not the right one as I have a customised tool on it). Therefore, I have tasks: `T_ROB_L`, `T_ROB_R`, `ROS_StateServer`, `T_GRIP_Motion`, `T_GRIP_Motion_L` and `T_GRIP_State` on the YuMi FlexPendant. They can be executed correctly, and I can see the left gripper try to do the calibration during starting.

Then on the Linux PC, I run `roslaunch yumi_launch yumi_pos_control.launch`. Note that I only add the part for the yumi_setup as without it the launch file can't be run correctly (refer to  #14 ). When I do:
`rostopic echo /joint_states` I get:

> header: 
>   seq: 1914
>   stamp: 
>     secs: 1516895022
>     nsecs: 857414007
>   frame_id: ''
> name: ['yumi_joint_1_r', 'yumi_joint_2_r', 'yumi_joint_7_r', 'yumi_joint_3_r', 'yumi_joint_4_r', 'yumi_joint_5_r', 'yumi_joint_6_r', 'yumi_joint_1_l', 'yumi_joint_2_l', 'yumi_joint_7_l', 'yumi_joint_3_l', 'yumi_joint_4_l', 'yumi_joint_5_l', 'yumi_joint_6_l', 'gripper_r_joint', 'gripper_r_joint_m', 'gripper_l_joint', 'gripper_l_joint_m']
> position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.010100000537931919, 0.010100000537931919]
> velocity: []
> effort: []

As you can see, the opening of the left gripper has the correct value (rviz also display this correctly). But I still only get all '0' value for the arms.

----

One thing I did but I am not sure if it's related:
I modify the tool
`https://github.com/kth-ros-pkg/yumi/blob/cb56067ea661670af179c179a02dd2197442be58/yumi_hw/rapid/RWS/ROS_motion_left.mod#L97
to
`MoveAbsJ target,move_speed,\T:=cycle_time,stop_mode,tool_gripper;`
where `tool_gripper` is the gripper/tool that I calibrated/identified using the YuMi program which calculates the mass and center of mass of the tool (see ABB manual).

I found this is useful to alleviate the issue that the motion supervision is trigger if the tool is relatively heavy. Note that I found this in the ABB_driver interface (yumi_support package), so not sure if it is still useful here.

Thank you so much for your help.

lz89, 2018-01-28:
I tried to launch the gazebo robot via `roslaunch yumi_launch yumi_gazebo_pos_control.launch`.
Except an error happened due to the unknow module `yumi_hw.srv` in:
https://github.com/kth-ros-pkg/yumi/blob/cb56067ea661670af179c179a02dd2197442be58/yumi_hw/scripts/gazebo_grippers_remap.py#L8
The rviz and gazebo seems runs correctly.
By doing `rostopic list` I can see the topics (i.e. `/yumi/joint_pos_controller_x_y/command`) to command the virtual robot (note that I don't see these topics in the real robot). The joint value from the `/joint_state` and `/yumi/joint_state` looks correct and the robot in rviz has correct initial pose.

diogoalmeida, 2018-01-29:
Hey @lz89, should those questions maybe be in different issues? I've tried to replicate your last question by running ``roslaunch yumi_launch yumi_gazebo_pos_control.launch`` but that led me to a new issue #21, on the most recent version of egm_modifications.

lz89, 2018-01-29:
Hi @diogoalmeida , yes, the gazebo should be a different issue. But here I just want to provide additional information to show that the YuMi in simulation is working (at least for the arm), but don't know why the real robot launch doesn't work.

Thank you.

lz89, 2018-02-11:
Yoshua kindly did a remote session with me try to figure what was the issue.
The solution was to reset the RAPID program and load the corresponding modules. However, the issue may happen from time to time, so that a backup of a working version for the YuMi controller is recommended. Before backup, you should calibrate the customised tool load and edit the line in the RAPID (as mentioned above) to ensure the MoveAbsJ function call the calibrated tool load.

When the issue occur, do a RAPID restart from the RobotStudio, then restore the working version of the system. 

It's still unknown why the issue can still occur, maybe Yoshua can comment.

Thanks very much.

closed 2018-02-11 by lz89

YoshuaNava, 2018-02-11:
Hey,
Glad to help :)

I believe it was because of the tool calibration. I'll try to add some information about custom tool calibration  soon.