kth-ros-pkg/yumi #18: Question: Control rate of yumi_hw (RWS) interface

lz89, 2018-01-26:
I wonder where I can set the control rate of this interface? I know the maximum is 10Hz for the RWS interface. I read through the code in yumi_hw package, and find the following.
The following function is called in a thread:

The `sampling_rate_` is set to 0.1, but it is not being set anywhere.
https://github.com/kth-ros-pkg/yumi/blob/cb56067ea661670af179c179a02dd2197442be58/yumi_hw/include/yumi_hw/yumi_hw_rws.h#L258

However, the control period in the `yumi_hw_ifce_node.cpp` is set to 1.0 second as shown:

https://github.com/kth-ros-pkg/yumi/blob/cb56067ea661670af179c179a02dd2197442be58/yumi_hw/src/yumi_hw_ifce_node.cpp#L129

I see that in the `YumiRapidInterface`, the spinOnce() of `industrial::message_manager::MessageManager` is called in each loop, as shown:

https://github.com/kth-ros-pkg/yumi/blob/cb56067ea661670af179c179a02dd2197442be58/yumi_hw/include/yumi_hw/yumi_hw_rws.h#L180

The document of `industrial::message_manager::MessageManager` states that:

> The message manager can be run in two ways. spin(), similar to ROS this executes a blocking execution indefinitely or spinOnce(), where a single execution of the manager is performed. In spinOnce mode, other server operations can be made, but the server application must make certain to **execute the spinOnce() function at a minimum rate so as not to loose connection data**.

So where this minimum rate is defined/configured, and is it 10Hz in this case?

Thank you.

diogoalmeida, 2018-03-28:
@lz89 , did you find an answer to this?

lz89, 2018-03-30:
@diogoalmeida Hi, I haven't found the answer yet. Hope Yoshua could give some hints. Thanks!

diogoalmeida, 2018-03-30:
Ok, @YoshuaNava is informed. Issue is that since we actually don't use RWS in our systems, it might be hard to come up with a straightforward answer. Maybe @jontje can help?

jontje, 2018-04-06:
Unfortunately I am not familiar with how the RWS interaction has been set up in this code. The `sampling_rate_` appears to have more to do with the ROS side of things, rather than the robot controller side.

diogoalmeida, 2018-05-09:
Closing this as it seems dead.

closed 2018-05-09 by diogoalmeida