kth-ros-pkg/yumi #21: Error launching yumi_gazebo_pos_control

diogoalmeida, 2018-01-29:
Hej, 

On the latest version of egm_modifications I get an undefined argument error:

```
 roslaunch yumi_launch yumi_gazebo_pos_control.launch
... logging to /home/diogo/.ros/log/0b3b5d90-01d6-11e8-a503-9cebe82fce71/roslaunch-diogo-dell-xps-4590.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
None Undefined substitution argument yumi_setup 
when evaluating expression 'yumi_setup == 'robot_centric''
XacroException(None,)
when processing file: /home/diogo/Documents/PhD/Research/yumi_catkin_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/diogo/Documents/PhD/Research/yumi_catkin_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro' arms_interface:=PositionJointInterface grippers_interface:=EffortJointInterface] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find yumi_description)/urdf/yumi.urdf.xacro' arms_interface:=$(arg arms_hardware_interface) grippers_interface:=$(arg grippers_hardware_interface)" name="robot_description"/>
The traceback for the exception was written to the log file
```

diogoalmeida, 2018-01-29:
Solved in 5281d39

closed 2018-01-29 by diogoalmeida

sinamr66, 2018-09-27:
Hi Diogo,

I am afraid that the problem is still there...

Best,
Sina

diogoalmeida, 2018-09-27:
When using the latest version of egm_modifications? I'll try to reproduce later today.

sinamr66, 2018-09-27:
Hi Diogo,

Yes, Actually tried it with the latest version and the version you just mentioned before...
Thx a lot

diogoalmeida, 2018-09-27:
I've set up a virtual machine, where I installed ROS-kinetic and the yumi workspace as described [in the wiki](https://github.com/kth-ros-pkg/yumi/wiki/Installation). I had an error related to the hector_xacro tools, which I fixed by running

```
$ sudo apt-get install ros-kinetic-hector-*
```

Can you confirm that installing those packages fixes the issue for you?