kth-ros-pkg/yumi #24: how use RWS to  control  Live robot

LeonRobot, 2018-03-19:
Hi  @YoshuaNava ,
First of all, let me thank you all for your works and the wiki! I was following the wiki to set-up the framework, however, when I read the section 'Live robot'(https://github.com/kth-ros-pkg/yumi/wiki/Live-robot), I don't find that how I should to do with the RWS, I think that you had miss that.

what's more,  I need to control the robot joint speed，but I read the Related Issue#7，find  that the EGM features  is no open source. So I want to know that if I can control the position and velocity of the waypoints in realtime to make up the trajectory I want.
I want to post a joint position command to the robot every 1 second and want to make the waypoint as smooth as possible without any pause. How should I do？Can you give me some advice？

Thanks!

Leon

YoshuaNava, 2018-03-21:
Hello @LeonRobot,
Thank you! 

I just added a section describing how to launch the nodes with RWS.

If you want to get a really smooth trajectory without using EGM, I think you have to send a good number of waypoints between start and goal. In [Documentation/Videos](https://github.com/kth-ros-pkg/yumi/wiki/Documentation#videos) I put two examples of that, from Orebro University.

@rtkg could you describe the setup you had for those experiments?

Hope this helps,
Yoshua

LeonRobot, 2018-03-22:
I am very glad to receive your reply. Your reply has helped me,  I know what to do.  In the section RWS, there is a mistake, I think the value of MAX_NUM_JOINTS should be 20.
By the way, I'm in China now, the Internet can't turn over the wall. I can't see the video you uploaded now. I'll watch these videos a few days later.

Thanks!
Leon

YoshuaNava, 2018-03-28:
Great to hear. Thanks for the feedback :)

closed 2018-03-28 by YoshuaNava