kth-ros-pkg/yumi #25: Confusion with Firmware Setup

juanluislm, 2018-03-27:
Hi,

I am trying to set up YuMi for position control, but the instructions in the wiki are a bit confusing.

We want to set up YuMi for position control, so we are using the files in the ABB_driver folder and in the Grippers folder. However, we find this part a bit confusing:

[image]

The last three tasks are "ALL RWS tasks". Does that mean that we have to set up the RWS tasks as well?

Also, we have a similar question regarding the grippers:

[image]

The last three tasks are "ALL T_GRIP_XXX tasks". Does that mean we have to attach those 3 scripts to all the gripper tasks individually?

Thanks,
Juan

YoshuaNava, 2018-03-28:
Hello @juanluislm,
The RWS interface tasks are: T_ROB_L, T_ROB_R, ROS_MotionServer_L, ROS_MotionServer_R, ROS_StateServer_L, ROS_StateServer_R.

The gripper tasks are: T_GRIP_Motion, T_GRIP_State, T_GRIP_Motion_L, T_GRIP_Motion_R.

You should add the corresponding ROS_common, ROS_messages and ROS_socket to each group of tasks. Keep in mind that each group (Grippers or ABB_driver) of files contains different definitions, even though they have the same name. We have kept the original names and structure of most of the files we got from ABB, specially for the ABB Driver and RWS control interfaces, which we use the least at KTH.

Hope this helps. Please let me know how it goes.

Yoshua

neubotech, 2018-03-29:
Hi @YoshuaNava,

I work with @juanluislm on setup.  Thank you for your prompt reply on the issue.  

It would be great if you can clarify few questions below:

1. Can Grippers, ABB_driver, and EGM load at the same time on YuMi?
2. If yes, from your reply, you are suggesting to create groups of tasks on RobotStudio.  We are not sure how to do that in the controller tab.  If a group task cannot be created, does it mean that we have to create an automatic port for each of the tasks?
3.   Does the KTH EGM module require EGM module purchased from ABB?
4.  We were able to get some feedback information from YuMi, but the YuMi state visualization on the MOVEIT! via Rviz is jittering.  I'm wondering if you have seen such problems, and has a solution to fix it?

Thank you so much for your time and patience!

Nan

YoshuaNava, 2018-04-10:
Hello @neubotech,
1. We normally load one control interface at the same time. Nonetheless, I think that you can load more than one if you create multiple "systems", each with their own set of RAPID files. [Link](http://developercenter.robotstudio.com/BlobProxy/manuals/RobotStudioOpManual/doc91.html)
3. Yes, the EGM license needs to be purchased from ABB.
4. Which control interface are you using? Could you send me some more info?

Hope this helps,
Yoshua

diogoalmeida, 2018-05-09:
Closing this as it seems dead.

closed 2018-05-09 by diogoalmeida

RYANZYC, 2019-01-14:
HI, I have a same problem with Firmware Setup. I am new to robotstudio.
there are abb_driver,grippers,and RWS rapid files in yumi_hw/rapid/ now. I want to use position control, It would be great if you can clarify few questions below:
1.If I can not have EGM files ,can I use RWS files to position control?
2.Should I load three groups of rapid file in robotstudio at the same time?
[image]
[image]
There are my Firmware Setups, is it right?Should T_ROB_L/T_ROB_L have both sys files of RWS path and abb_driver path?There are also two ROS_motion_left/right.mod files in RWS and abb_driver,should I load in T_ROB_L/T_ROB_L task at the same time? If yes , whether should I change the name of same name's file?
THANK YOU!

diogoalmeida, 2019-01-14:
Hi @RYANZYC. I never set up the firmware nor ever used robotstudio. I remember vaguely that you should be able to use RWS for position control, but everything else you've asked is a bit too much for me to reply.

We currently do not have anyone at KTH assigned to maintain this repository, and as such it might be difficult for you to find better answers here. I'll come back to you if this changes, but don't count on it.

Good luck,
Diogo

RYANZYC, 2019-01-14:
Hi @diogoalmeida . Do you mean that you have not maintain the package of yumi ? I want to use ros to control active robot. I learn the wiki that sets up firmware， but I have some confuse about this ，should I write  a new issue?

diogoalmeida, 2019-01-14:
I mean that at its current state, we are not actively developing the package, and use it internally within the limits of what is described on the wiki.

If you are willing to contribute to our package, e.g., by providing instructions on alternative use cases, you're welcome to do it, but you shouldn't count on getting much support in configuring the code for use cases that are not currently covered on our wiki.

RYANZYC, 2019-01-14:
Maybe I didn't express my meaning. All I use is provided by yumi package.but I have some confusion about wiki tutorial of Firmware[https://github.com/kth-ros-pkg/yumi/wiki/Firmware](url). , so I want to ask the specific details.My problem is above.

muh-ahmed, 2020-09-15:
@YoshuaNava, please help me regarding a firmware setup
After creating the tasks for ABB driver-position control, how to add the scripts to the tasks as described in https://github.com/kth-ros-pkg/yumi/wiki/Firmware ?

YoshuaNava, 2020-10-01:
@muh-ahmed From what I remember you have to load the scripts using ABB RobotStudio.