kth-ros-pkg/yumi #26: How to link one file to multiple tasks when setting up the firmware

Gene0213, 2018-03-27:
Hi @YoshuaNava ,

I am a little bit confused when setting up the firmware of Yumi following your instruction. We decide to do position control through ABB driver and created tasks followed the table you provided. The question is how to link one file to multiple tasks (for example, how to link "ABB_driver/ROS_common.sys" to "All RWS tasks" )?

Best,
Gene

YoshuaNava, 2018-03-28:
Hello @Gene0213,
Due to our robot being booked atm, I can only look into this starting on Friday. In the meantime, could you send me the output of `rostopic list` when you run the YuMi nodes on the physical robot?

YoshuaNava, 2018-04-10:
@Gene0213 
Are you still having this problem?

Gene0213, 2018-04-10:
Dear Yushua,

Sorry for the late reply. Our Yumi is serving for another project now. I
cannot test it lately. I will give you feedback once I have a chance to
test it. Thank you for your time.

Best,
Jinfa

closed 2018-08-22 by diogoalmeida

leishi07, 2018-10-01:
Hi, Yoshua,

I have exactly the same problem. I modified the controller.yaml file, ran abb driver. Trajectory is executed in rviz but real robot doesn't move. Moveit always does a fake execution of trajectory. Here are the results of rostopic list:
/attached_collision_object
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/joint_states
/left_arm/feedback_states
/left_arm/joint_path_command
/left_arm/joint_states
/left_arm/joint_trajectory_action/cancel
/left_arm/joint_trajectory_action/feedback
/left_arm/joint_trajectory_action/goal
/left_arm/joint_trajectory_action/result
/left_arm/joint_trajectory_action/status
/left_arm/robot_status
/move_group/cancel
/move_group/display_contacts
/move_group/display_cost_sources
/move_group/display_grasp_markers
/move_group/display_planned_path
/move_group/fake_controller_joint_states
/move_group/feedback
/move_group/goal
/move_group/monitored_planning_scene
/move_group/motion_plan_request
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/recognized_object_array
/right_arm/feedback_states
/right_arm/joint_path_command
/right_arm/joint_states
/right_arm/joint_trajectory_action/cancel
/right_arm/joint_trajectory_action/feedback
/right_arm/joint_trajectory_action/goal
/right_arm/joint_trajectory_action/result
/right_arm/joint_trajectory_action/status
/right_arm/robot_status
/rosout
/rosout_agg
/rviz_lei_Latitude_5580_30600_3478918543834961267/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz_lei_Latitude_5580_30600_3478918543834961267/motionplanning_planning_scene_monitor/parameter_updates
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/tf
/tf_static
/trajectory_execution_event
/yumi/gripper_l_effort_cmd
/yumi/gripper_r_effort_cmd
/yumi/gripper_states

Any ideas?

Best Regards,

Lei

leishi07, 2018-10-02:
It is working now. After I ran through the launch files, I found out that I made a stupid mistake. I didn't pay attention to the difference between demo.launch and demo_online.launch.

hsaleem1, 2020-09-15:
Hi @leishi07 

I have done the firmware setup as provided in this repository. Then, I also modified controller.yaml file and ran abb driver.
The robot in rviz is following the trajectory but the real robot is not moving. I also checked the "rostopic list", and it is publishing the topic just like yours.
Can you please help me out?

Thanks.

Mahdiford, 2022-06-27:
> Hi @leishi07
>
> I have done the firmware setup as provided in this repository. Then, I also modified controller.yaml file and ran abb driver. The robot in rviz is following the trajectory but the real [...]

Did you solve this problem?