kth-ros-pkg/yumi #27: EGM RAPID modules

nash169, 2018-03-29:
Hi,
I'm trying to configure the ABB YuMi (IRB14000) with EGM support and I would like use your package to control it. We (EPFL) got from ABB the ".rlf" license file for EGM and I activated it through RobotStudio. Where can I get the related RAPID modules (or any other necessary files)? Can you release these files or should I ask ABB for them as well?

Thank you very much!

gavanderhoorn, 2018-03-29:
(I'm not at KTH, but involved in the development of the EGM support packages)

We're working on releasing those pieces of infrastructure right at this moment. This will either be done this week, or next week.

I expect to be able to provide more information soon.

nash169, 2018-03-29:
Hi gavanderhoorn,
thank you for your quick reply! Looking forward to hearing from you soon.

closed 2018-03-29 by diogoalmeida

gavanderhoorn, 2018-04-05:
Just to let you know: see the following repositories for the egm and rws libraries that were used (in older versions) by the packages in this repository:

 - [ros-industrial/abb_librws](https://github.com/ros-industrial/abb_librws)
 - [ros-industrial/abb_libegm](https://github.com/ros-industrial/abb_libegm)

Note again: there may be incompatibilities between the packages in this repository (ie: `kth-ros-pkg`) and the two I link above, but that is a result of the development process.

I envision that now that these two libraries have been officially open-sourced by ABB the consumers of those libraries (that had early access to them) will gradually converge towards these and abandon the early-access versions.

That may take some time though.

gavanderhoorn, 2018-04-05:
O, I just noticed that the RAPID side of things was not included yet.

Perhaps @jontje can provide some clarifications here.

diogoalmeida, 2018-04-06:
Thanks for the update @gavanderhoorn!

jontje, 2018-04-06:
The EGM and RWS libraries linked above primarily provide communication functionality for _**general use cases**_. When RAPID code and robot controller configurations are involved, then it becomes more of a _**specific application**_.

My intention is to, at least initially, create some guidelines for robot controller configurations, and then also provide a few samples for how to use the libraries (and also containing RAPID code). More complete setups could also be included later.

nash169, 2018-04-11:
I'm sorry guys, I got confused. I just started to work with yumi so I'm a kind of noob in this stuff. We got the EGM license because we needed to control the robot at a higher frequency than 10Hz. In order to use this package ("egm_modifications" branch) I thought it was necessary to have on the robot side the egm rapid modules as shown here https://github.com/kth-ros-pkg/yumi/wiki/Firmware. I've been following this guide http://wiki.ros.org/abb/Tutorials/InstallServer for setting up the robot for control via ros. If I'm wrong can anyone explain me what I'm supposed to do for the configuration of the robot? Is there any other tutorial or guide I missed?

nash169, 2018-04-13:
Hi guys,
If I can use the EGM without the dedicated RAPID modules can anyone explain me how to do that? Instead, if these modules are mandatory is there a way to get them? Should I contact ABB directly?

diogoalmeida, 2018-04-13:
My understanding of what @jontje wrote is that a _specific application_ entails a particular robot, in this case Yumi. So you'd need some RAPID code to get to communicate over EGM, but I'm not aware of details. I'm pretty sure we need to run some RAPID code to get the communication going.

nash169, 2018-04-13:
Hi Diogo,
I agree with you but I don't know where to find these modules. Looking at the tutorial under the "firmware" section of the wiki I guess I'd need these files:
EGM/TRobEGM.mod
EGM/TRobMain.mod
EGM/TRobSynchronization.sys
EGM/TRobSG.sys

We got the EGM license from the ABB and I already installed it through RobotStudio but it's not clear to me what to do now without the EGM modules. Would you be able to tell me the configuration that you set up through RobotStudio (I mean "Task" and "Automatic loading of Modules")?

diogoalmeida, 2018-04-13:
I am really the wrong person to discuss that, as I work with the ROS side of things. @YoshuaNava might be able to point you on the right directions, otherwise @jontje is the man you're looking for. If you're lacking those files, I'd assume ABB can provide them.

nash169, 2018-04-13:
Hi jontje,
Thank you for your reply. So am I supposed to write down the EGM Rapid modules myself? Why you can't provide them?

YoshuaNava, 2018-04-13:
Hello,
@jontje
We at KTH have an old edition of libegm and librws, with the corresponding RAPID modules. Could we share those modules? So that people have a guide on how to launch EGM and communicate with the computer?

It would be just as an example, as we're gonna support the new versions of libegm and librws. The development tasks will take place soon.

jontje, 2018-04-16:
As I mentioned earlier, I am currently working on some ROS sample nodes for the newest libraries, which includes RAPID code. They are not ready at the moment, but I hope to have something later this month.

@YoshuaNava: You can share the old versions' RAPID modules, but please don't put them on a public repo. I would prefer not to spread the old versions too much.

@nash169: If you have the time, then I suggest that you write a short RAPID program to familiarize yourself with the different steps for EGM, as well as the different arguments that exists. Then test the program with RobotStudio simulations, I think that it would be beneficial in the long run.

nash169, 2018-04-17:
Thank you guys for the help.
I'll try to get more familiar with the RAPID code. However having some sample code to look at would be very useful. @YoshuaNava if you can't put these modules on a public repo can you send them to me by email (bernardo.fichera@epfl.ch)?

nash169, 2018-04-23:
Hi @YoshuaNava,
Could you send me the old EGM Rapid modules?

robberthofmanfm, 2021-02-11:
Are there already any RAPID examples available for EGM? 
I understand that we can learn more about EGM through `EGMGetId` up until `EGMReset` as @jontje carefully outlined, but at this moment it's not possible to reach a working EGM system through exclusively following the wiki (as mentioned earlier, the [Firmware](https://github.com/kth-ros-pkg/yumi/wiki/Firmware) part refers to files that are not publicly available).

gavanderhoorn, 2021-02-11:
Not a direct answer to your question @robberthofmanfm, but the new [abb_robot_driver](https://github.com/ros-industrial/abb_robot_driver) fully supports YuMi, provided you have the EGM license on your controller.

The State Machine addin written by @jontje makes setting things up rather simple.

ROS services (via RWS) are provided for starting, stopping and otherwise coordinating everything remotely from ROS.

robberthofmanfm, 2021-02-11:
Thanks a lot for the link, I'll spend some time exploring that repo now.
I already installed the state machine add-in as it's marked _optional_ of the [libegm README](https://github.com/ros-industrial/abb_libegm#overview), but finding the exact EGM steps that jontje outlined above in there confused me even more.

Thank you both for the fantastic work, too! It's much appreciated.

AshenOneZHM, 2022-10-25:
@YoshuaNava @nash169 Excuse me, I'm now studying how to control Yumi through the EGM in the code package you provided. I checked the issues related to egm, and thought that if I want to use some demo of ros provided in your code package to control Yumi, I still need the related egm rapid program. I see the problem you discussed long ago, and it seems that the old version of the rapid code package related to egm can be provided, Can you send it to me? If you can, I would appreciate it (a request from a humble graduate student). Email: 1063136772@qq.com