kth-ros-pkg/yumi #28: Is there a way to directly publish Cartesian space command?

LeonRobot, 2018-05-09:
Hi All,
First of all, let me thank you all for your works!
I find that this package can only issue joint motion commands. I think there is a way to use the YUMI's own inverse kinematics solution, so that you don't have to write your own robot inverse kinematics program. Some time ago I used ros moveit's kinematics solution API, but I found that the solution is often not optimal or there are no solution. So I want to know how you  do inverse kinematics or whether there is an interface that can directly publish Cartesian space positions command.
Thanks in advance for our help.
Best
Leon

diogoalmeida, 2018-05-09:
Hi Leon,

As far as I am aware, you should compute the desired joint space commands yourself. The easiest option to do this, as far as I am aware, is to use [KDL's inverse kinematics solvers](http://docs.ros.org/hydro/api/orocos_kdl/html/classKDL_1_1ChainIkSolverPos__NR.html) (alternatively you have [joint velocity solvers](http://docs.ros.org/hydro/api/orocos_kdl/html/classKDL_1_1ChainIkSolverVel__pinv.html)).

More advanced solutions might work better if you wan't to make sure joint and/or workspace constraints are respected. In this case you might want to take a look at [hqp based solutions](https://github.com/OrebroUniversity/hiqp).

Best,
Diogo

closed 2018-05-09 by diogoalmeida

gavanderhoorn, 2018-05-09:
Wouldn't a big limitation (with EGM at least) be that there is no support for Cartesian motion with YuMi at the moment?

diogoalmeida, 2018-05-09:
Right :)