kth-ros-pkg/yumi #30: abb_rws_interface files are missing

Maxpad, 2018-08-02:
Hi All,

First of all I would like to thank you for your work and the wiki!
I was following the wiki in order to run the Yumi's camera, yet, I failed to run the rws_continuous_trigger.cpp

I have searched and found that the rws_interface_yumi.h file and all the abb_rws_interface repository are missing.

Could you please help me?

diogoalmeida, 2018-08-02:
Hi Max!

@YoshuaNava (or maybe @jontje, if he has the time) will be able to give a definitive answer, but as far as I am aware, we are using a proprietary version of the `abb_rws_interface` package. There is an open source version [here](https://github.com/ros-industrial/abb_librws), but I think our (KTH) yumi package is still not ready to use it.

@YoshuaNava, is this correct?

Maxpad, 2018-08-03:
Thank you for your answer, I will check that.

closed 2018-08-22 by diogoalmeida

Martinpanzy, 2019-05-14:
Hi, I have the same question. And I am wondering whether it has been fixed or not. Thanks!