kth-ros-pkg/yumi #33: Query!!!

Sinchiguano, 2018-12-21:
Hi,
I am a master student from CVUT in The Czech Republic, we got the Yumi robot recently, I was wondering if we could use the job you guys did lately. If so, I then got a another question, do I really need to clone the package (git clone git@gits-15.sys.kth.se:RPL-Internal/abb_experimental.git  ## You need to set up your KTH-git public key first)?

Thanks for your answer!
Cesar

diogoalmeida, 2018-12-21:
Hi Cesar,

You're very welcome to use our yumi packages. However, the abb_experimental package we have at our private repositories are proprietary and we are not allowed to share them. ABB has released that package, however, as part of the ROS industrial project. You can find it [here](https://github.com/ros-industrial/abb_experimental). 

Please note that we can't guarantee that everything will work "out of the box" with that package, but I encourage you to try. If minor changes are required to get our package to work with the open-source abb-experimental package, I'd be very thankful if you could submit a pull request with those changes :). 

Good luck!
Diogo

YoshuaNava, 2018-12-21:
Hi @Sinchiguano,
Thank you for your interest in our package.
As @diogoalmeida indicated, we can't share our abb_experimental package because it's proprietary. Furthermore, you'd need to buy a license from ABB if you want to do velocity control (i.e. use EGM) on YuMi.

Before leaving KTH I did some work towards supporting the open source ABB [RWS](https://github.com/ros-industrial/abb_librws) and [EGM](https://github.com/ros-industrial/abb_libegm) libraries, which you can find here:

[abb_irb14000_core](https://github.com/kth-ros-pkg/abb_irb14000_core)
[abb_irb14000_description](https://github.com/kth-ros-pkg/abb_irb14000_description)

The launchfiles, description and code are cleaner than in this package. RWS support should be up and running. What's missing is supporting the new EGM library and other features.

As Diogo said, we would be super thankful if you use them and then find time to contribute back your changes.

All the best,
Yoshua

Sinchiguano, 2018-12-21:
Thanks both of you guys for such a nice and fast answer. 

Best regards
Cesar

closed 2019-01-14 by diogoalmeida