kth-ros-pkg/yumi #37: ikfast for yumi

RYANZYC, 2019-04-21:
Hello maintainer!I want to use ikfast instead of KDL algorithm to solve the kinematics of the manipulator.But when I follow the tutorial, the following prompt appears:

[image]
[image]
Have you configured the ikfast kinematics algorithm for yumi?

diogoalmeida, 2019-04-22:
Hi,

I am not aware of anyone having configured ikfast for yumi at our lab. You might want to ask for help at the [openrave](https://github.com/rdiankov/openrave) repository page.

Best,
Diogo

PS: in future issues, consider using `` ``` `` blocks to copy and paste terminal output, like this:
```
# This is a test
$ cd /home/user/Documents
``` 

More information on that [here](https://help.github.com/en/articles/creating-and-highlighting-code-blocks)

closed 2019-04-22 by diogoalmeida

muh-ahmed, 2020-09-16:
Hi @RYANZYC,
Can you please guide if you are succeeded to use IKFast instead of KDL algorithm to solve the kinematics of the Yumi?

Thanks.