kth-ros-pkg/yumi #40: Controller of YuMi 14050

Gloriabhsfer, 2019-06-11:
Hi All,

I am searching online for packages of YuMi 14050, the single arm version of YuMi. It seems like most packages for YuMi is about irb14000.  Can I still use packages for 14000 to control the single arm, or I need to create my own model in gazebo and use ros_control to create my own packages?

Thank you
Gloria

diogoalmeida, 2019-06-12:
Hi Gloria,

The code in this repository should largely overlap what you would need to control the irb14050. Unfortunately, we never came around making the code more modular and adaptable to other ABB robots, so I am not sure on how to proceed on that front. 

Best,
Diogo

Gloriabhsfer, 2019-06-15:
Hi Diogo:

Thanks for your reply. I will close this issue since I get my answer.

Best,
Gloria

closed 2019-06-15 by Gloriabhsfer