kth-ros-pkg/yumi #43: Failed to control IR14000 using ROS/ABB_driver #33

amranov, 2019-06-21:
Hi guys ,
We are trying to control the real robot with ABB interface we followed  tutorial [](https://github.com/kth-ros-pkg/yumi/wiki/Firmware) to install the firmware, (We don't have the EGM modules). when we launch : roslaunch yumi_launch yumi_pos_control.launch we get this error :
```
[ WARN] [1561104675.666918206]: Send failure, no callback support
[ INFO] [1561104675.666949402]: Connection failed, attempting reconnect
[ERROR] [1561104675.667712026]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
```
Any suggestions?

YoshuaNava, 2019-06-21:
Hi @amranov,
Can you give us more information about your setup? Computer, network, ROS version, firmware on the robot, if you have seen any errors on the flexpendant, etc.

amranov, 2019-06-25:
Hi @YoshuaNava 
Sorry for the late reply, i am using ThinkPad-E460 , 7.3 GB memory
Processor :Intel® Core™ i7-6498DU CPU @ 2.50GHz × 4 
Graphics :Intel® HD Graphics 510 (Skylake GT1), 64 bits
ROS kinetic, 16.04
i have no error on the flexpendant.
connect an Ethernet cable from my computer to the robot's XP23 port as mentioned in the tutorial.
FIrmware  RobotWare Services (RWS) - Position control and  ABB Driver - Position control, since we don't have the EGM.
when i ping into the robot ip it works normally 
but when i try run `roslaunch yumi_support robot_interface.launch` i got this error : 
```
[ WARN] [1561104675.666918206]: Send failure, no callback support
[ INFO] [1561104675.666949402]: Connection failed, attempting reconnect
[ERROR] [1561104675.667712026]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
```
the confusing thing is when i `rostopic echo /joint_states` it seems like the topic is publishing normally 
- Also, i would like to know if its possible to control Yumi without having EGM? (Sorry but i am quite beginner)

jpsilverio, 2019-09-23:
Hi guys,

Was there any good solution to this issue? I have exactly the same problem: a YuMi robot without EGM, trying to run `roslaunch yumi_support robot_interface.launch` and getting 

`[ERROR] [1569241937.637927847]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)`
`[ERROR] [1569241937.638078138]: Failed to receive message length`
`[ERROR] [1569241937.638148697]: Failed to receive incoming message`

I am also using kinetic and Ubuntu 16.04. I can ping 192.168.125.50 without a problem.

Thanks!
João

closed 2019-10-24 by diogoalmeida

hsaleem1, 2020-09-15:
Hi @amranov, @jpsilverio 
Did you guys solved these issues? I am having the same problem.

Thanks.

Mahdiford, 2022-06-17:
I have the same problem. 

p.s. I have  EGM license. I can run example 3 [here](https://github.com/ros-industrial/abb_robot_driver/blob/master/abb_robot_bringup_examples/README.md) without any errors.