kth-ros-pkg/yumi #44: Issues with joint states, and EGM timeout

jbeck28, 2019-09-06:
I've put a fair bit of work into getting this code to work on an IRB1200, with much success. I'm specifically interested in getting EGM to work with ros_control. A few quirks have shown up, and I believe my limited skill in c++ might be getting in the way. 

first, in my RVIZ view, I occasionally get all-zeros joint state to pop up. This is mostly a nuisance, since moveit sometimes tries to use those joint states as the start state. I checked the publishers to the /joint_states topic and found that both irb1200_joint_state_publisher, and /irb1200_hw were to publishing. (note that irb1200_hw is the equivalent of your yumi_ifce_node, with the main control loop in it). I'm not certain if this is the typical operation, but I understand that the XXXX_joint_state_publisher is managed by the controller_manager, and is therefore part of ros_control? 

The other issue I'm having is that if I start EGM and don't send it a command relatively quickly, it appears as if the robot disconnects (I'm using velocity control), and stops publishing the loop period. Jontje has alluded to this being due to the egm_controller_interface expecting references in reply to each request for feedback. I'm curious if this behavior is something you've noticed, and have a fix, or if you've any idea where to point me? 

I can post code if need be, but it's very much the same as this package but some numbers (joints per arm, number of arms) changed, and names changed from YUMI to IRB1200. 

Thanks in advance,
Josh

diogoalmeida, 2019-09-07:
Hi Josh,

Thank you for your contact. I am not familiar enough with the code to help you directly, but maybe we can get somewhere. 

```
I occasionally get all-zeros joint state to pop up
```
In your `ifce` equivalent, do you get zeros when calling the `read` method? Our equivalent is found on [this line](https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_hw/src/yumi_hw_ifce_node.cpp#L164). It calls the [getCurrentJointStates](https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_hw/src/yumi_hw_egm.cpp#L177) method and something could be going wrong there. Maybe you're calling it faster than EGM is able to provide you with information?

```
if I start EGM and don't send it a command relatively quickly, it appears as if the robot disconnects (I'm using velocity control), and stops publishing the loop period
```
When we call the [write method](https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_hw/src/yumi_hw_ifce_node.cpp#L170), we enforce joint limits and call the egm_common `setJointVelTargets` and `setJointPosTargets` ([here](https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_hw/src/yumi_hw_egm.cpp#L505)). Are you doing something similar? It's hard to be concrete without seeing your code.

Best,
Diogo

jbeck28, 2019-09-09:
First I'd like to thank you for being so prompt in your reply. 

In regards to the robot ceasing to respond after a certain period of time. The solution I've arrived at which works every time is to immediately execute a path as soon as the driver begins to display the control loop period [https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_hw/src/yumi_hw_ifce_node.cpp#L172](here), which prevents the robot falling into unresponsiveness. These messages stop being printed when the robot stops responding. 

To clarify, my code and your code are quite literally identical, besides changing YuMi to IRB1200. I did however, setup the transmissions in my URDF, and the controller config myself, perhaps that's a source of error? 

I will print the joint positions to the terminal directly from the read method that you mentioned, and see if I'm getting zeros from the robot controller. Perhaps the fact the robot goes through the network rather than being wired directly to my computer...

diogoalmeida, 2019-09-10:
I doubt that the URDF would be the cause of your error, but adjustments in the controller config might. 

```
Perhaps the fact the robot goes through the network rather than being wired directly to my computer...
```
I'm not familiar enough with EGM to know what happens if you query it too fast. 

It appears that in our latest change we [commented out the line that tries to ensure a steady communication rate](https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_hw/src/yumi_hw_ifce_node.cpp#L174). Maybe this is the source of the issue? 

If you uncomment it and set the control rate to be reasonable (~200Hz seems to be the best we can get with our Yumi's), will you still get the zeros?

jbeck28, 2019-09-11:
Still working on the timeout problem, but the zero joint states were being published courtesy of me not commenting out [https://github.com/kth-ros-pkg/yumi/blob/4520af912c0ab3532225d84fed2cabe4870379bd/yumi_launch/launch/yumi_traj_vel_control.launch#L49](this). Though I'm not sure why anything was being published since yumi/joint_states did not have any publishers itself... I have found that the messages don't exist anymore.

diogoalmeida, 2019-10-24:
Closing this issue. Feel free to reopen it if anything changes.

closed 2019-10-24 by diogoalmeida