kth-ros-pkg/yumi #45: How to do grasps with MoveIt!'s Pick & Place pipeline?

xabigarde, 2019-09-09:
Hi,

Thanks for mantaining this great repo for working with the ABB YuMi Robot. I hope this is the right place to ask this question.

I was developing a simple demo with C++, and when trying to use MoveIt's Pick & Place pipeline ([following its corresponding tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pick_place/pick_place_tutorial.html)) I began to wonder if it is even possible to use this pipeline with the current implementation of this repo's packages, considering the YuMi's grippers control is limitted to the effort controllers for the time being (as stated [here](https://github.com/kth-ros-pkg/yumi/wiki/Grippers-control)).

After searching extensibly for a solution, I came to the same conclusion that user _gvdhoorn_ explains [here ](https://answers.ros.org/question/198706/how-to-plan-moves-with-gripper-attached-to-arm/)answering to a somewhat related question:

> For actual control I've seen two approaches being taken most of the time:
> 
> 1. separate gripper control from arm control: use MoveIt planning to get the EEF to a desired location. Now use your gripper services or action server directly to control it.
> 2. define proper end-effector groups in your MoveIt configuration package, complete with associated controllers and have MoveIt plan trajectories for your EEF and arm.

So far I've done what the first option describes, but I'd much rather use the built in capabilities that MoveIt! has for grasping objects.

If using only the Gazebo simulator, I could venture onto the second option. However, for using the real YuMi robot, the second option would require implementing the joint position controller for the gripper fingers, with the corresponding RAPID services for the IRC5 controller. This goes way beyond my current knowledge and skills.
Which brings me to my current situation of not knowing what the best possible workaround for the second option is. Any hints?

Also: I have recently found out that [these yumi demos](https://github.com/OrebroUniversity/yumi_demos) from OrebroUniversity use the [HiQP framework](https://github.com/OrebroUniversity/hiqp) by Marcus A. Johansson for grasp planning and control instead of MoveIt!. However, the documentation for using this framework is minimal, and its development and maintenance seems to be halted, so I have therefore not explored it myself and would rather avoid using it as it is.

I'd very much appreciate any guidance towards a solution to this problem. 
Precisely, how do the maintainers of this repo (or others using it like me) implement the grasps control and Pick & Place planning?

If the only solution would require extending the current effort controller (so we can set the fingers to a particular opening posture, instead of just opened/closed), I would of course be willing to work on that under the maintainers' guidance and assistance.

diogoalmeida, 2019-09-10:
Hi @xabigarde thanks for your message!

I have no experience with the MoveIt's Pick&Place pipeline. However, I might be able to partially help you. 

> However, for using the real YuMi robot, the second option would require implementing the joint position controller for the gripper fingers, with the corresponding RAPID services for the IRC5 controller. This goes way beyond my current knowledge and skills.
Which brings me to my current situation of not knowing what the best possible workaround for the second option is. Any hints?

I have recently collaborated in a project where we had to deal with the gripper, and the effort control interface was not cutting it. Simply put, the gripper does not send feedback on the position while moving, so it is impossible to close a feedback loop to control its position. However, you can modify the RAPID code to send position commands directly, rather than effort. This will send the gripper to a given position. 

You would have to [modify this file](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_hw/rapid/Grippers/GripperMotion_right.mod). Instead of
```
Hand_GripInward \holdForce:=grasp_force;
```
we just used the 'grasp_force' value as an intended aperture, which we sent to the ``Hand_MoveTo`` command. 

In summary, we hacked the RAPID code to use the effort command to the gripper as a de facto position command. This is not ideal, but saved us the time in learning how to define EGM messages. Furthermore, we edited the code directly on the flexpendant, which is a headache. 

Once you make these changes, it should be possible to more easily integrate the Pick&Place pipeline by forwarding a trajectory's position values to the gripper, using a feedforward controller from ros control. 

> Also: I have recently found out that these yumi demos from OrebroUniversity use the HiQP framework by Marcus A. Johansson for grasp planning and control instead of MoveIt!. However, the [...]

Unfortunately, that is the case with many open sourced scientific projects. I have seen that framework being used and it does the job nicely, but you'll have to use its source code as 'documentation', I believe. If you want to get a better understanding of the HiQP as a concept, I can recommend [this article](https://ieeexplore.ieee.org/iel5/8860/5975532/05766760.pdf) as a pretty straightforward description of the framework and [this other article](https://journals.sagepub.com/doi/pdf/10.1177/0278364914521306) for the computational tricks required to make it fast.

xabigarde, 2019-09-10:
Thanks a lot for your fast reply @diogoalmeida !! 

I'll  give it a shot at editing the RAPID code as you suggest. 

> I have no experience with the MoveIt's Pick&Place pipeline.

If not by use of MoveIt! Pick&Place pipeline, how have you performed the grasps control? By implementing it yourselves in a more straightforward manner? 

I'm still interested in how others using the YuMi with its default grippers are managing the grasp sequences, since we must all be on the same boat with the effort controllers.

diogoalmeida, 2019-09-10:
Grasping is an unsolved problem, so I would expect many people to implement their own algorithms to, e.g., find a desired grasping pose, and them have a separate pipeline to reach this pose and execute the grasp. 

With our YuMis, I think the most common approach is to open the gripper with maximum effort, reach the grasp pose and then close the gripper with maximum effort, that is, no position control whatsoever.

xabigarde, 2019-09-10:
Great, thanks! I was on the right track then.
I'll try your suggested RAPID "hack" in the following days and report back.
Should I close this issue?

closed 2019-09-10 by xabigarde