kth-ros-pkg/yumi #46: Is this package private or public?

wongearth, 2019-10-04:
I got stuck at installation procedure below. Just wondering if this package is private or public?

git clone git@gits-15.sys.kth.se:RPL-Internal/abb_experimental.git
Cloning into 'abb_experimental'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

diogoalmeida, 2019-10-04:
Hi,

As with most open source repositories, this software is provided 'as is', and any support you get is done so freely and on the contributors' spare time. Please take that in consideration when reporting issues.

That said, the package you mention contains proprietary code from ABB required to interface with EGM. If you purchased EGM from ABB they will provide you with the code. Otherwise, you can safely disregard that topic, and use the 'ABB' or 'RWS' drivers provided in this repo, for position control of yumi's joints. The wiki should provide enough information on those control methods.

closed 2019-10-04 by diogoalmeida

wongearth, 2019-10-05:
Hi...

Thank you for your help. I am new to Github. Sorry for my mistake.

Wong