kth-ros-pkg/yumi #47: Using Yumi with ROS + MoveIt!

beevabeeva, 2019-11-29:
We have managed to get as far as having MoveIt (+ rviz) get the correct current position of the robot. However, when trying to execute a path to a goal position, the robot does not move.

In addition, there are no gripper topics listed. MoveIT complains about not having the gripper states:
[image]

We followed the instructions (after digging around in other branches to find the required backups (addition-demo, two-grippers realtime) ). We found someone with a similar issue
(https://github.com/OrebroUniversity/yumi/issues/21) but there were no detailed steps listed to solve the issue.

Any help would be appreciated.

diogoalmeida, 2019-12-02:
Hi,

Which robot driver are you using?

beevabeeva, 2019-12-11:
Hi @diogoalmeida,

we have tried both the ABB driver and the RWS driver.
We do not have an EGM license so we can't try that. If needed, we can try get it.

diogoalmeida, 2020-04-27:
Completely missed this reply, apologies for that.

For future reference, we do not provide support for setting up your Yumi systems, but all our information on the topic can be found on the [wiki](https://github.com/kth-ros-pkg/yumi/wiki).

closed 2020-04-27 by diogoalmeida