kth-ros-pkg/yumi #48: Installation error

yanjunyang, 2019-12-01:
I have a problem with building the workspace with catkin_make. I am using ROS melodic within python 2.7.16

```
yanjun_yang@MU00160572L:~/yumi_depends_ws$ catkin_make -DCMAKE_BUILD_TYPE=Release
Base path: /home/yanjun_yang/yumi_depends_ws
Source space: /home/yanjun_yang/yumi_depends_ws/src
Build space: /home/yanjun_yang/yumi_depends_ws/build
Devel space: /home/yanjun_yang/yumi_depends_ws/devel
Install space: /home/yanjun_yang/yumi_depends_ws/install
####
#### Running command: "cmake /home/yanjun_yang/yumi_depends_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/yanjun_yang/yumi_depends_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yanjun_yang/yumi_depends_ws/install -G Unix Makefiles" in "/home/yanjun_yang/yumi_depends_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yanjun_yang/yumi_depends_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/yanjun_yang/yumi_depends_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/yanjun_yang/yumi_depends_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yanjun_yang/yumi_depends_ws/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.15") 
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.17
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 26 packages in topological order:
-- ~~  - abb (metapackage)
-- ~~  - abb_irb2400_moveit_config
-- ~~  - abb_irb6640_moveit_config
-- ~~  - industrial_core (metapackage)
-- ~~  - industrial_deprecated
-- ~~  - industrial_msgs
-- ~~  - abb_irb2400_support
-- ~~  - abb_irb5400_support
-- ~~  - abb_irb6600_support
-- ~~  - abb_irb6640_support
-- ~~  - simple_message
-- ~~  - gazebo_mimic
-- ~~  - industrial_utils
-- ~~  - industrial_robot_client
-- ~~  - abb_driver
-- ~~  - industrial_robot_simulator
-- ~~  - abb_irb2400_moveit_plugins
-- ~~  - industrial_trajectory_filters
-- ~~  - yumi_cameras
-- ~~  - yumi_control
-- ~~  - yumi_description
-- ~~  - yumi_hw
-- ~~  - yumi_moveit_config
-- ~~  - yumi_support
-- ~~  - yumi_launch
-- ~~  - yumi_test_controllers
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'abb'
-- ==> add_subdirectory(abb/abb)
-- +++ processing catkin package: 'abb_irb2400_moveit_config'
-- ==> add_subdirectory(abb/abb_irb2400_moveit_config)
-- +++ processing catkin package: 'abb_irb6640_moveit_config'
-- ==> add_subdirectory(abb/abb_irb6640_moveit_config)
-- +++ processing catkin metapackage: 'industrial_core'
-- ==> add_subdirectory(industrial_core/industrial_core)
-- +++ processing catkin package: 'industrial_deprecated'
-- ==> add_subdirectory(industrial_core/industrial_deprecated)
-- +++ processing catkin package: 'industrial_msgs'
-- ==> add_subdirectory(industrial_core/industrial_msgs)
[... 1291 lines cut ...]
 inline bool getSoftJointLimits(urdf::JointConstSharedPtr urdf_joint, SoftJointLimits& soft_limits)
             ^~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/joint_limits_interface/joint_limits_urdf.h:84:13: note:   no known conversion for argument 1 from ‘const boost::shared_ptr<const urdf::Joint>’ to ‘urdf::JointConstSharedPtr {aka std::shared_ptr<const urdf::Joint>}’
yumi/yumi_hw/CMakeFiles/yumi_hw.dir/build.make:62: recipe for target 'yumi/yumi_hw/CMakeFiles/yumi_hw.dir/src/yumi_hw.cpp.o' failed
make[2]: *** [yumi/yumi_hw/CMakeFiles/yumi_hw.dir/src/yumi_hw.cpp.o] Error 1
CMakeFiles/Makefile2:7959: recipe for target 'yumi/yumi_hw/CMakeFiles/yumi_hw.dir/all' failed
make[1]: *** [yumi/yumi_hw/CMakeFiles/yumi_hw.dir/all] Error 2
/home/yanjun_yang/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In destructor ‘virtual gazebo::MimicPlugin::~MimicPlugin()’:
/home/yanjun_yang/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:48:18: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(this->updateConnection);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yanjun_yang/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In member function ‘virtual void gazebo::MimicPlugin::UpdateChild()’:
/home/yanjun_yang/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:88:42: error: ‘class gazebo::physics::Joint’ has no member named ‘GetAngle’; did you mean ‘GetForce’?
     mimic_joint_->SetPosition(0, joint_->GetAngle(0).Radian()*multiplier_);
                                          ^~~~~~~~
                                          GetForce
yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/build.make:62: recipe for target 'yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/src/gazebo_mimic_plugin/mimic_plugin.cpp.o' failed
make[2]: *** [yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/src/gazebo_mimic_plugin/mimic_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:4181: recipe for target 'yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/all' failed
make[1]: *** [yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```
What should I do to fix this problem? Thanks a lot!

diogoalmeida, 2019-12-02:
Hi,

Make sure to switch the package to the 'melodic' branch.

In the package directory run: 

```
$ git checkout melodic
```

yanjunyang, 2019-12-02:
> Hi,
> 
> Make sure to switch the package to the 'melodic' branch.
> 
> In the package directory run:
> 
> ```
> $ git checkout melodic
> ```

Thanks for your reply! I didn't notice that. Besides, I couldn't find EGM/TRobEGM.mod, EGM/TRobMain.mod, EGM/TRobSynchronization.sys and EGM/TRobSG.sys in yumi_hw/rapid/

diogoalmeida, 2019-12-02:
No problem! 

The EGM files require you to purchace the EGM module from ABB, which we cannot provide in this repository. You might want to try the ABB or RWS driver options.

Best,
Diogo

closed 2019-12-02 by diogoalmeida

yanjunyang, 2019-12-02:
> No problem!
> 
> The EGM files require you to purchace the EGM module from ABB, which we cannot provide in this repository. You might want to try the ABB or RWS driver options.
> 
> Best,
> Diogo

Do you mean that these files are with the purchased EGM module together? Once we get the EGM, where should we get these files?

Cheers
Yanjun Yang

diogoalmeida, 2019-12-02:
I believe so. Those files should be provided by ABB, and they should provide with support to your institution. Unfortunately, I have not dealt with ABB directly myself, so I cannot be sure of what the process to acquire the files should be.

yanjunyang, 2019-12-02:
> I believe so. Those files should be provided by ABB, and they should provide with support to your institution. Unfortunately, I have not dealt with ABB directly myself, so I cannot be sure of what the process to acquire the files should be.

Thanks so much for your kindly help!

jontje, 2019-12-02:
> The EGM files require you to purchace the EGM module from ABB

Just as a clarification, those files are not included in the EGM license. The latest available version of them are included in the free RobotWare [StateMachine Add-In](https://robotapps.robotstudio.com/#/viewApp/7fa7065f-457f-47ce-98d7-c04882e703ee).

Though I do not know if the KTH implementation is compatible with the latest files.

@yanjunyang, you can find more details how to install the Add-In in the [user manual](https://robotapps.blob.core.windows.net/appreferences/docs/2093c0e8-d469-4188-bdd2-ca42e27cba5cUserManual.pdf), if you are interested.

diogoalmeida, 2019-12-02:
Thanks for chiming in @jontje!

yanjunyangmonash, 2020-11-25:
Hi @diogoalmeida , I recently got a new laptop and tried to reinstall this, but I got the problem again. I noticed that using `sudo apt-get install protobuf-compiler` will install 3.0.0 version, but I have to use 3.11.4 for the protobuf. Therefore I omit this one command

Then I got the error like this
```
CMake Error at yumi/gazebo_mimic/CMakeLists.txt:8 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/yanjun/yumi_depends_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/yanjun/yumi_depends_ws/build/CMakeFiles/CMakeError.log".
```
I just want to know do I have to use 3.0.0 version to make it?

diogoalmeida, 2020-11-25:
Hi!

The error seems to be related to your system missing gazebo. That shouldn't be related to protobuf. If you followed the installation instructions and are still having issues, please try to run

sudo apt-get install ros-kinetic-gazebo-ros-pkgs

Best,
Diogo

yanjunyangmonash, 2020-11-26:
> Hi! The error seems to be related to your system missing gazebo. That shouldn't be related to protobuf. If you followed the installation instructions and are still having issues, please try to run [...]

Hi Diogo

Thanks for your quick reply, since I still use ROS melodic, I modified your command to `sudo apt-get install ros-kinetic-gazebo-ros-pkgs`. Besides, The gazebo package will automatically download protocbuf3.0.0, so I removed 3.11.4 version to avoid conflict. While I still fail to make it and the problem seems to come from industrial_core. Hope you have time to go through the log

```
Base path: /home/yanjun/yumi_depends_ws
Source space: /home/yanjun/yumi_depends_ws/src
Build space: /home/yanjun/yumi_depends_ws/build
Devel space: /home/yanjun/yumi_depends_ws/devel
Install space: /home/yanjun/yumi_depends_ws/install
####
#### Running command: "cmake /home/yanjun/yumi_depends_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/yanjun/yumi_depends_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yanjun/yumi_depends_ws/install -G Unix Makefiles" in "/home/yanjun/yumi_depends_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yanjun/yumi_depends_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/yanjun/yumi_depends_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/yanjun/yumi_depends_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yanjun/yumi_depends_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 25 packages in topological order:
-- ~~  - abb (metapackage)
-- ~~  - abb_irb2400_moveit_config
-- ~~  - abb_irb6640_moveit_config
-- ~~  - industrial_core (metapackage)
-- ~~  - industrial_deprecated
-- ~~  - industrial_msgs
-- ~~  - abb_irb2400_support
-- ~~  - abb_irb5400_support
-- ~~  - abb_irb6600_support
-- ~~  - abb_irb6640_support
-- ~~  - simple_message
-- ~~  - gazebo_mimic
-- ~~  - industrial_utils
-- ~~  - industrial_robot_client
-- ~~  - industrial_robot_simulator
-- ~~  - abb_irb2400_moveit_plugins
-- ~~  - industrial_trajectory_filters
-- ~~  - yumi_cameras
-- ~~  - yumi_control
-- ~~  - yumi_description
-- ~~  - yumi_hw
-- ~~  - yumi_moveit_config
-- ~~  - yumi_support
-- ~~  - yumi_launch
-- ~~  - yumi_test_controllers
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'abb'
-- ==> add_subdirectory(abb/abb)
-- +++ processing catkin package: 'abb_irb2400_moveit_config'
-- ==> add_subdirectory(abb/abb_irb2400_moveit_config)
-- +++ processing catkin package: 'abb_irb6640_moveit_config'
-- ==> add_subdirectory(abb/abb_irb6640_moveit_config)
-- +++ processing catkin metapackage: 'industrial_core'
-- ==> add_subdirectory(industrial_core/industrial_core)
-- +++ processing catkin package: 'industrial_deprecated'
-- ==> add_subdirectory(industrial_core/industrial_deprecated)
[... 1575 lines cut ...]
[ 75%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message.dir/src/messages/joint_feedback_message.cpp.o
[ 76%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message.dir/src/messages/joint_traj_pt_message.cpp.o
[ 76%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/joint_feedback.cpp.o
[ 76%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/joint_traj_pt.cpp.o
[ 77%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/joint_traj_pt_full.cpp.o
[ 77%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/joint_traj.cpp.o
[ 78%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/robot_status.cpp.o
[ 78%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/messages/joint_message.cpp.o
[ 78%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message.dir/src/messages/joint_traj_pt_full_message.cpp.o
[ 79%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/messages/joint_feedback_message.cpp.o
[ 79%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/messages/joint_traj_pt_message.cpp.o
[ 79%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/messages/joint_traj_pt_full_message.cpp.o
[ 79%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message.dir/src/messages/robot_status_message.cpp.o
[ 80%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/messages/robot_status_message.cpp.o
[ 81%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message.dir/src/simple_comms_fault_handler.cpp.o
[ 81%] Building CXX object industrial_core/simple_message/CMakeFiles/simple_message_bswap.dir/src/simple_comms_fault_handler.cpp.o
[ 81%] Linking CXX shared library /home/yanjun/yumi_depends_ws/devel/lib/libsimple_message.so
[ 81%] Built target simple_message
[ 82%] Linking CXX shared library /home/yanjun/yumi_depends_ws/devel/lib/libsimple_message_bswap.so
[ 82%] Built target simple_message_bswap
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

```

diogoalmeida, 2020-11-26:
That's a compilation issue with the [https://github.com/OrebroUniversity/industrial_core](https://github.com/OrebroUniversity/industrial_core) package. It looks very deprecated, so it might not work with melodic. You might want to see if the [ros industrial version](https://github.com/ros-industrial/industrial_core) works (replace the `industrial_core` package from `yumi_depends_ws/src` and clone the new one. 

If that doesn't solve it, I'm afraid I cannot help you.

yanjunyangmonash, 2020-11-26:
Yes, I tried the whole day with different combinations, and finally worked with `git clone -b kinetic https://github.com/ros-industrial/industrial_core.git`, `git clone https://github.com/NITTC-Robotics/yumi.git`. I also tried to replace the simple_message folder as the author modified max message size for yumi, but that doesn't work. 

I also have another question, do you still use it or there are better packages for yumi? Thanks for your time!