kth-ros-pkg/yumi #50: How to do  kinematic teaching on YuMi?

YantianZha, 2020-02-04:
Hello, I am writing to ask how to do  kinematic teaching on a YuMi robot (basically by manually moving robot's arm and meanwhile record the trajectory)? Thanks!

On Baxter robot there are codes to [record](https://github.com/RethinkRobotics/baxter_examples/blob/master/scripts/joint_recorder.py) and [play back](https://github.com/RethinkRobotics/baxter_examples/blob/master/scripts/joint_trajectory_file_playback.py) trajectories. While I can try to adapt the code to YuMi robot, but I suspect it's not that straightforward? Maybe I also need to handle something with EGM interface?

gavanderhoorn, 2020-02-05:
Cross-posted to ROS Answers: [How to do Kinesthetic teaching on ABB YuMi?](https://answers.ros.org/question/343375).

YantianZha, 2020-02-05:
Yes.

jontje, 2020-02-07:
> Hello, I am writing to ask how to do Kinesthetic teaching on a YuMi robot (basically by manually moving robot's arm and meanwhile record the trajectory)? Thanks!

@YantianZha, I have done something similar before, and I used YuMi's `leadthrough` functionality in combination with `Externally Guided Motion` (EGM).

It might also be possible to use [Robot Web Service](http://developercenter.robotstudio.com/webservice/api_reference) (RWS), instead of EGM, but then the recorded trajectories won't have the same granularity. And you would probably need to do the teaching very slowly (because RWS communication is much slower than EGM).

> I also need to purchase the Externally Guided Motion licence from ABB, right?

It depends on how "smooth" you need the recorded trajectories to be, and in addition, you will need to at least use RobotWare `6.09`. I would recommend EGM if you want the recorded the motion to be as close as possible to the teaching.

> On Baxter robot there are codes to [record](https://github.com/RethinkRobotics/baxter_examples/blob/master/scripts/joint_recorder.py) and [play [...]

If you use EGM, then it might be easiest just to wrap two [EGMContollerInterface](https://github.com/ros-industrial/abb_libegm/blob/89791927dd25c99df0344fa6e23279579868bf38/include/abb_libegm/egm_controller_interface.h#L46-L60) instances (from the [abb_libegm](https://github.com/ros-industrial/abb_libegm) library) inside a tiny ROS node. And then you can customize how you record the trajectories.

And in the YuMi's RAPID program you could use `EGMStreamStart`/`EGMStreamStop`, and `SetLeadThrough \On`/`SetLeadThrough \Off` instructions.

YantianZha, 2020-02-08:
@jontje Thanks for your information!

I found [this approach](https://github.com/kth-ros-pkg/yumi/wiki/Lead-Through-mode) under kth-ros-yumi is probably a simpler way to record demo trajectories.

closed 2020-02-13 by diogoalmeida