kth-ros-pkg/yumi #53: launch file is not running

Ahmedali952, 2020-04-25:
I need to simulate YuMi robot in gazebo and failed to launch position trajectory control of joints by using command: roslaunch yumi_launch yumi_gazebo_traj_pos_control.launch

 **Error**: resource not found: hector_xacro_tools
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/ahmed/yumi_depends_ws/yumi_depends_ws/src
ROS path [2]=/home/ahmed/yumi_depends_ws/src
ROS path [3]=/opt/ros/kinetic/share None
when processing file: /home/ahmed/yumi_depends_ws/src/yumi/yumi_description/urdf/Grippers/yumi_servo_gripper.xacro
included from: /home/ahmed/yumi_depends_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/ahmed/yumi_depends_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro' arms_interface:=PositionJointInterface grippers_interface:=EffortJointInterface yumi_setup:=default] returned with code [2].

Param xml is <param command="$(find xacro)/xacro.py '$(find yumi_description)/urdf/yumi.urdf.xacro' arms_interface:=$(arg arms_hardware_interface) grippers_interface:=$(arg grippers_hardware_interface) yumi_setup:=$(arg yumi_setup)" name="robot_description"/>
The traceback for the exception was written to the log file

diogoalmeida, 2020-04-25:
Hi,

The error message is clear, "resource not found: hector_xacro_tools" you might try to install it with "sudo apt install ros-melodic-hector-xacro-tools".

Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/ahmed/yumi_depends_ws/src/yumi/yumi_description/urdf/yumi.urdf.xacro' arms_interface:=PositionJointInterface grippers_interface:=EffortJointInterface yumi_setup:=default] returned with code [2].

Param xml is

Ahmedali952, 2020-04-25:
@diogoalmeida, thank you for your response. Now the error is removed by installing it from "sudo apt install ros-melodic-hector-xacro-tools".

closed 2020-04-25 by Ahmedali952

YoshuaNava, 2020-04-25:
For future reference, the wiki has instructions on how to install all the dependencies of this package.