kth-ros-pkg/yumi #56: Unable to connect to live yumi robot

SnowWhite0808, 2020-07-27:
I am using Ubuntu 18.04.4 LTS with ROS Melodic and I am trying to connect to the live Yumi robot.

I followed this tutorial https://github.com/kth-ros-pkg/yumi/wiki/Firmware to install the firmware, (We dont have the EGM modules). 
And also finished the installation and netwok setup following the wiki ( https://github.com/kth-ros-pkg/yumi/wiki/Installation ).
But when I launch : roslaunch yumi_support robot_interface.launch or  roslaunch yumi_launch yumi_pos_control.launch, I get this error :

[ERROR] [1595792296.959756397]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ WARN] [1595792296.959765686]: Not connected, bytes not sent
[ERROR] [1595792296.959772055]: Failed to receive message length
[ERROR] [1595792296.959778484]: Failed to receive incoming message
[ WARN] [1595792296.959784695]: Send failure, no callback support

Could anyone give me some advice? Any possible reasons for this will be helpful for me.
Thanks in advance!

ArFuchs, 2020-08-10:
Hi, 

I'm also trying to connect to Yumi since a few days. I added the necessary tasks and files and the FlexPendant says its connected to Yumi, but in the terminal in which I run "roslaunch yumi_support robot_interface.launch" I get the same error as SnowWhite0808. 
I followed all the steps in the wiki, but it doesn't work. 

I hope someone can help, because it's really frustrating.
Thanks in advance.

muh-ahmed, 2020-09-10:
@ArFuchs, can you please guide me that I am doing setup for live connection to Yumi.
I followed the instructions of https://github.com/kth-ros-pkg/yumi/wiki/Firmware, but when I came to point 15 "Once this is done, click Next then Apply", then the following error came:
failed to get mastership for the sytem 
release mastership on system and try again.

Please help.
Thanks.

HauNan, 2020-10-15:
@ArFuchs Hi,ArFuchs, Knowing that your yumi runs well and congratulations.
Would you please tell me if you install the EGM modules or not? I'm curious if I call the T-CAMERA functions in ROS, is that necessary for me to purchase the EGM modules.
Now I could drive yumi by abb-driver but without the camera.