kth-ros-pkg/yumi #57: hector_xacro_tools

ritalaezza, 2020-07-31:
Hej,

I want to setup a YuMi in our lab at Chalmers, and I am looking to your repo as a starting point. 

What exactly is the purpose of **hector_xacro_tools**, in  [yumi/yumi_description/urdf/Grippers/yumi_servo_gripper.xacro](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_description/urdf/Grippers/yumi_servo_gripper.xacro)? Why isn't it included in dependencies?

diogoalmeida, 2020-07-31:
Hi Rita,

You should be able to install the missing package with

```bash
sudo apt install ros-melodic-hector-xacro-tools
```

Make sure to replace `melodic` with the correct ROS version you're using. If I'm not wrong, this package has utilities that help in parsing a xacro file into a urdf file that ROS can use. The reason for using xacro is that it facilitates building robot descriptions in a more modular way.

I am not sure why it isn't listed as a dependency. It might be that it should come with some other dependency, and is omitted due to that, or it might be that we should list it in the package manifesto, but I can't know for sure without being at my computer.

Hope this helps!

ritalaezza, 2020-08-02:
Hi Diogo,

Thank you for your fast reply :) 

I am using ROS Melodic as well, and have installed the package. If you look at the original repo: [hector_xacro_tools](https://github.com/tu-darmstadt-ros-pkg/hector_models/tree/kinetic-devel/hector_xacro_tools), perhaps you can understand which components are used. It don't think it provides any additional xacro parsing functionalities, but I may be mistaken.

Best,
Rita