kth-ros-pkg/yumi #58: YuMi urdf xacro

ritalaezza, 2020-08-02:
For the gazebo launch files to work I had to modify the [yumi.urdf.xacro](https://github.com/kth-ros-pkg/yumi/blob/melodic/yumi_description/urdf/yumi.urdf.xacro) file, since the variable `yumi_parent` is defined after it is used. This is how the file looks with my changes:

```
...
<xacro:property name="yumi_setup" value="$(arg yumi_setup)" />
<xacro:if value="${yumi_setup == 'robot_centric'}">
    <!-- Pure YuMi URDF, with no joints connecting it to any other robot or pedestal. Neither accessories. -->
</xacro:if>
<xacro:if value="${yumi_setup == 'default'}">
    <xacro:property name="yumi_parent" value="world" />
    <link name="world"/>
    <joint name="world_joint" type="fixed">
        <parent link="world"/>
        <child link="yumi_base_link"/>
        <origin xyz="0 0 0.1" rpy="0 0 0" />
    </joint>
</xacro:if>

<!--yumi-->
<xacro:yumi name="yumi" hardware_interface="hardware_interface/$(arg arms_interface)" parent="${yumi_parent}" >
    <origin xyz="0 0 0.1" rpy="0 0 0" />
</xacro:yumi>

<!--right gripper-->
<xacro:yumi_servo_gripper name="gripper_r" hardware_interface="hardware_interface/$(arg grippers_interface)" parent="yumi_link_7_r">
    <origin xyz="0 0 0.007" rpy="0 0 ${PI}" />
</xacro:yumi_servo_gripper>
...
```

Also, if `yumi_setup` is set to "robot_centric" in the launch files,  `yumi_parent` is left undefined. Finally, I'm not sure if the variable `PI` is even  used.

ps. do you have any file with guidelines for contributing to your repo?

kappa95, 2021-02-12:
I don't know if it is useful for you but I have recently modified the urdf.xacro file in order to add a table fixed on the bottom of the Yumi using directly the part of `<xacro:if value="$....">` using a different setup. In this way: 
```xml
  <xacro:if value="${yumi_setup == 'fede_base'}">
    <joint name="robot_base_joint" type="fixed">
      <parent link="base"/>
      <child link="yumi_base_link"/>
      <origin xyz="-${base_width/2} 0 ${base_height/2}" rpy="0 0 0"/>
    </joint>
  </xacro:if>
```

robberthofmanfm, 2021-02-18:
I just experienced the same trouble as ritalaezza. After finding [the right xacro command on the wiki](https://github.com/kth-ros-pkg/yumi/wiki/URDF#how-to-export-yumis-urdf-from-the-xacro-files) with all the parameters:
`rosrun xacro xacro yumi.urdf.xacro arms_interface:=VelocityJointInterface grippers_interface:=EffortJoinInterface yumi_setup:=robot_centric -o yumi.urdf` 
It indeed gave me the follow error:
```
name 'yumi_parent' is not defined 
when evaluating expression 'yumi_parent'
when processing file: yumi.urdf.xacro
```
Just like ritalaezza, my solution was to add the following line in `yumi.urdf.xacro`, just after the ``comment:
`<xacro:property name="yumi_parent" value="world" />`
I'm not sure if `world` is the right value, but at least I can produce a `yumi.urdf` file.

On a different note, it might be nice to also define default values for the parameters passed, e.g.
 ```
<xacro:arg name="arms_interface" default="VelocityJointInterface"/>
<xacro:arg name="grippers_interface" default="EffortJointInterface"/>
<xacro:arg name="yumi_setup" default="robot_centric"/>
```

Another thing is that [on the wiki](https://github.com/kth-ros-pkg/yumi/wiki/URDF#how-to-export-yumis-urdf-from-the-xacro-files), it says `EffortJoinInterface`, not `EffortJointInterface`. I think that's a typo, which is consequently also present in [`yumi.urdf`](https://github.com/kth-ros-pkg/yumi/blob/egm_modifications/yumi_description/urdf/yumi.urdf#L742)

kappa95, 2021-02-18:
I have notice that EffortJoinInterface works. It is not a typo. Actually i modified successfully the URDF adding a table following the guide.

Federico Insero

robberthofmanfm, 2021-02-25:
When launching `gazebo.launch`, with `EffortJoinInterface` (with typo), the following errors occur:
```
[FATAL] [1614246805.489533072, 0.349000000]: No matching hardware interface found for 'hardware_interface/EffortJoinInterface' while loading interfaces for gripper_r_joint
[FATAL] [1614246805.489605294, 0.349000000]: Could not initialize robot simulation interface
```

When correcting the typo to `EffortJointInterface`, these errors disappear.

`[ INFO] [1614247350.404103390, 0.152000000]: Loaded gazebo_ros_control.`

kappa95, 2021-02-25:
Hi,
I'm using ROS Kinetic probably is a typo specific of that ROS version (?!).
Each time I'm using this package I encounter different issues.

Federico Insero

robberthofmanfm, 2021-03-02:
I've got another, different comment that relates to the xacro files.
On the [wiki](https://github.com/kth-ros-pkg/yumi/wiki/URDF#how-to-export-yumis-urdf-from-the-xacro-files), it says to run `rosrun xacro xacro yumi.urdf.xacro -o yumi.urdf` (with some parameters), but I've experienced that this is often not enough. You'll also need to source your overlay (e.g. `source ~/catkin_ws/devel/setup.bash`) before running the command. Otherwise it will produce a URDF based on what's living in `/opt/ros/ROSVERSION/share/yumi/yumi_description/`.

Can we update the wiki so that it includes the overlay source, or at least warns about this?