kth-ros-pkg/yumi #62: Problems with the reading of the joints

kappa95, 2020-10-03:
Hi community,
I have a big issue in my project because the readings of the joints are wrong for the left arm.
[image]
[image]

On the shell are shown the results of the 2 commands: rostopic echo /left_arm/joint_states/position and rostopic echo /right_arm/joint_states/position 

I compared to the results of the jogging:
[image]
[image]

Can you help me please???

diogoalmeida, 2020-10-03:
Hi!

If you want help with software issues, it is helpful if you indicate which commands you have run and what's the terminal output. I would also recommend that you paste the output here, within a block code (using ``` to delimit the code should work).

For example, the rostopic commands you mentions would be pasted like

```
rostopic echo /left_arm/joint_states/position 
rostopic echo /right_arm/joint_states/position
```

Without knowing which commands you issued to see those results, it is not possible to help you. 

Best,
Diogo

kappa95, 2020-10-03:
Thank you so much Diogo!

In any case i solved the issues with the simulator and the reading of the joints using the solution of the pull requests on this repository: [https://github.com/kth-ros-pkg/yumi/pull/35](url)

closed 2020-10-03 by kappa95