kth-ros-pkg/yumi #66: Support for Ubuntu 18

robberthofmanfm, 2021-02-05:


robberthofmanfm, 2021-02-05:
Hi all,

I have successfully compiled all packages in this repository on a Ubuntu 20.04 + ROS Noetic setup.
Below, I'll outline the steps to reproduce this in hope to save someone in a similar situation some time.
I have not tested this thoroughly, so your success may still vary.
Here we go:

`git clone https://github.com/kth-ros-pkg/yumi.git`
in your favorite `catkin_ws/src/`

Try building already using `catkin build` or `catkin_make_isolated` but brace yourself for error messages.
Try to `sudo apt install ros-noetic-PACKAGE` for every PACKAGE you encounter in the errors that catkin produces.

You'll also need abb_librws, so go ahead and clone that into your `catkin_ws/src/`:
`git clone https://github.com/ros-industrial/abb_librws.git`

After a while, you'll probably reach the moment `catkin` tells you it cannot find `simple_message`. This is where the fun starts, since that requires an installation of [ros industrial](http://wiki.ros.org/Industrial/Install), which is does not have a binary install for Ubuntu 20.04 (i.e. you can't just `apt install` it). Building from source is a nice try, but does not work either, catkin will complain about some `trajectory_filters` that are missing. It's in hard times you get to know your heroes, in my case `simonschmeisser`, who wrote a patch on ROS industrial core that makes it compatible with ROS noetic. You can find it here:
https://github.com/simonschmeisser/industrial_core/tree/noetic-compat
So instead of cloning rosindustrial's source to your favorite `catkin_ws/src`, clone Simon's repo in `catkin_ws/src` and checkout in the right branch:
`git clone https://github.com/simonschmeisser/industrial_core.git`
`git checkout noetic-compat`
`git pull origin noetic-compat`

Now at least a different class of errors comes up, and they refer to actual code, of which the errors look something like the excerpt in the comment above. My favorite one is the following:
`yumi_hw.cpp:186:89: error: conversion from ‘urdf::JointConstSharedPtr’ {aka ‘std::shared_ptr<const urdf::Joint>’} to non-scalar type ‘const boost::shared_ptr<const urdf::Joint>’ requested
  186 |   const boost::shared_ptr<const urdf::Joint> urdf_joint_sitffness = urdf_model->getJoint(joint_name + std::string("_stiffness"));`

First thing I noticed is `urdf_joint_sitffness`. That must be a typo, right? However, if you search for use of variables that start with `urdf_joint_s`, this is the only occurence in the entirety of the kth-ros-pkg/yumi repo. So I don't know if it matters, but I changed the variable name to `urdf_joint_stiffness`. Same errors. 

Then I came across [this ros answers post](https://answers.ros.org/question/366073/urdfmodel-boostshared_ptrurdflink-boostshared_ptrurdfjoint/), which suggested swapping out the `boost::` types for `std::` types. I'm not sure what boost exactly is, but [this guy](https://stackoverflow.com/questions/6322245/should-i-switch-from-using-boostshared-ptr-to-stdshared-ptr) recommends just using `std::`. This was the last fix I had to do, so replace `boost` with `std` on line 185 and 186 in `ws_moveit/src/yumi/yumi_hw/src/yumi_hw.cpp`.

imitsioni, 2021-02-08:
Thanks for the update! I will close this issue and add a link to the wiki.

closed 2021-02-08 by imitsioni

robberthofmanfm, 2021-02-08:
Hey @imitsioni , what about the `urdf_joint_sitffness` ? Is that a typo? Should I open a pull request to correct it? Where is this variable used?

imitsioni, 2021-02-08:
As far as I can tell, it is not used somewhere on our side of the code. However, this seems to be related to the files used for the physical robot, not the simulated one.  Are you connecting to a real yumi?

robberthofmanfm, 2021-02-08:
Yes, but I haven't tested this code on the real yumi yet. If I find anything interesting related to joint stiffness I'll comment again here.

imitsioni, 2021-02-08:
Thank you! I am suspecting that this is not used anywhere and commenting it/removing it will not affect anything but I do not have access to the robot right now and cannot test it. If you see something breaking please let me know.

yanjunyangmonash, 2021-02-11:
Hi @robberthofmanfm, I'd like to know if your method works for ROS melodic + Ubuntu18.04? I tried combinations:

1. `git clone https://github.com/kth-ros-pkg/yumi.git`, `git clone https://github.com/simonschmeisser/industrial_core.git`, `catkin build`. It gives the error:
`/home/yanjun/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In destructor ‘virtual gazebo::MimicPlugin::~MimicPlugin()’:
/home/yanjun/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:48:18: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(this->updateConnection);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yanjun/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In member function ‘virtual void gazebo::MimicPlugin::UpdateChild()’:
/home/yanjun/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:88:42: error: ‘class gazebo::physics::Joint’ has no member named ‘GetAngle’; did you mean ‘GetForce’?
     mimic_joint_->SetPosition(0, joint_->GetAngle(0).Radian()*multiplier_);
                                          ^~~~~~~~
                                          GetForce
make[2]: *** [CMakeFiles/gazebo_mimic_plugin.dir/src/gazebo_mimic_plugin/mimic_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_mimic_plugin.dir/all] Error 2
make: *** [all] Error 2`

2. `git clone -b melodic https://github.com/kth-ros-pkg/yumi.git`, `git clone https://github.com/ros-industrial/abb_librws.git`, `git clone https://github.com/ros-industrial/industrial_core.git`, `catkin build`. The compiler shows couldn't ABB RWS library not found.

Would you mind provide some help?

robberthofmanfm, 2021-02-11:
Oh yes, forgot to mention. I removed the `gazebo_mimic` package. I haven't gotten that to compile.

robberthofmanfm, 2021-02-12:
It appears that I was too early to celebrate. Compilation did indeed succeed, but at runtime there are still some problems.
A first step in resolving these, is to - before deleting the `gazebo_mimic` package - install its dependencies:
`rosdep install --from-paths src/yumi/gazebo_mimic --ignore-src --rosdistro noetic`

yanjunyangmonash, 2021-02-14:
> It appears that I was too early to celebrate. Compilation did indeed succeed, but at runtime there are still some problems.
> A first step in resolving these, is to - before deleting the `gazebo_mimic` package - install its dependencies:
> `rosdep install --from-paths src/yumi/gazebo_mimic --ignore-src --rosdistro noetic`
> 
> For me, that resulted in the equivalent of:
> 
> ```
> sudo apt install ros-noetic-ros-control
> sudo apt install ros-noetic-ros-controllers
> ```

Hi, thanks for your reply. I already removed that folder but still got warnings like this. Will this warning cause any problem?
[image]

robberthofmanfm, 2021-02-14:
Yes, I suspect this will result into trouble later on.
I am not an expert myself, but I would try to `catkin build abb_librws` first and if that succeeds `catkin build` the whole workspace.

yanjunyangmonash, 2021-02-15:
> Yes, I suspect this will result into trouble later on.
> I am not an expert myself, but I would try to `catkin build abb_librws` first and if that succeeds `catkin build` the whole workspace.

I tried to follow your instruction but it still shows the error message. The reason for Yumi_cameras giving this error is because it looks for abb_rws_interface instead of abb_librws, which we cloned before. I replaced abb_rws_interface with abb_librws in both Cmakelist and package.xml, but couldn't compile the package and it seems I have to modify rws_continuous_trigger.cpp under `yumi_cameras/src`. Do I miss anything?

robberthofmanfm, 2021-02-15:
Ah yes, now that I rebuild the packages after deleting my `build` directory, I too have these warnings. I have only gotten these packages to compile, not to work at runtime, maybe that's why.

It indeed seems that to make this work, one needs to convert the `abb_rws_interface` to `abb_librws`. I have no idea what changed between those two and no idea how to migrate from one to the other. 

However, in [this issue](https://github.com/kth-ros-pkg/yumi/issues/27#issuecomment-777635627), we were pointed to [a new ABB robot driver](https://github.com/ros-industrial/abb_robot_driver). We're also trying to figure out how that works, but depending on what you're trying to do, it might be better to switch to that repository.

robberthofmanfm, 2021-02-15:
@imitsioni Maybe it's better to remove this issue from the wiki. It does compile with warnings, but I doubt we'll make it work with a physical yumi any time soon.

imitsioni, 2021-02-15:
You might be right, I will keep an eye out and try to test these changes after February.

YoshuaNava, 2021-02-21:
Hi @robberthofmanfm 

Thank you very much for your feedback and interest in the package.

I'm not working with YuMi anymore, but as developer of the package I would be happy to review any Pull Requests you can submit.

Btw, Kinetic is at end-of-life, http://wiki.ros.org/Distributions and Melodic will only have two more years. Does it make sense to rather implement support for Noetic and Ubuntu 20.04?

Best,
Yoshua

robberthofmanfm, 2021-02-22:
Hi @YoshuaNava,

I am always a fan of supporting latest versions and I am running 20.04 + Noetic myself. However, I don't see a way to ensure compatibility for all Ubuntu/ROS versions. PRs that make Noetic compatible will probably break compatibility with 16.04 + Kinetic (and maybe 18.04 + Melodic).

At this moment, we are using [ros-industrial/abb_robot_driver](https://github.com/ros-industrial/abb_robot_driver) to control YuMi over EGM, and using this repository mostly for the moveit configurations, yumi descriptions and launch files.
We will probably always use those two packages together, so we are not going to invest time in making this package work in a stand alone way (without a dependency to `abb_robot_driver`).

Kind regards,
Robbert

Mahdiford, 2022-06-28:
> @robberthofmanfm I have solved the problem with **yumi_hw** like this [#74 (comment)](https://github.com/kth-ros-pkg/yumi/issues/74#issuecomment-947960186), and it works with me on ROS Noetic.

@YoshuaNava @robberthofmanfm  @jontje @diogoalmeida I have tested this, and it worked fine on Noetic. The only thing that remained was about EGM. I have an EGM license. When I set up the robot in robot studio, the related EGM "*.mod" and "*.sys" files were downloaded under T_ROB_L and T_ROB_R tasks. While "yumi_traj_pos_control.launch" works fine with RWS, for the EGM, I got this error "YuMi EGM interface not available. RobotHW node will be stopped now." It cannot find "TRobEGM.sys." It seems in the new update, and the TRobEGM.sys file had become TRobEGM.mod.
Also, I tried "TRobEGM.sys" with no success. I am so confused about the EGM settings and what files and signals I should set up to be able to use EGM. 
Could you please provide some instructions/ clarification for setting up the EGM?

diogoalmeida, 2022-06-28:
> Could you please provide some instructions/ clarification for setting up the EGM?

Hey @Mahdiford, unfortunately, I am unable to help you with this. Are you able to get support from ABB?

robberthofmanfm, 2022-06-28:
We're not using `yumi_hw` but [`abb_robot_driver`](https://github.com/ros-industrial/abb_robot_driver). On that repo, you can find a lot of guidance on how to set up EGM. Some of them will probably also be needed to get `yumi_hw` working. I'm thinking about e.g. setting up these UDPUC devices on the YuMi side: https://github.com/ros-industrial/abb_robot_driver/tree/master/abb_egm_hardware_interface#troubleshooting 

On the YuMi side, `abb_robot_driver` expects https://robotapps.robotstudio.com/#/viewApp/c163de01-792e-4892-a290-37dbe050b6e1  to be installed.

Mahdiford, 2022-06-28:
> > Could you please provide some instructions/ clarification for setting up the EGM?
> 
> Hey @Mahdiford, unfortunately, I am unable to help you with this. Are you able to get support from ABB?

Hi Thanks, @diogoalmeida, ABB stated that they are not supporting the driver and they released it as is. I will try my chance.

> We're not using `yumi_hw` but [`abb_robot_driver`](https://github.com/ros-industrial/abb_robot_driver). On that repo, you can find a lot of guidance on how to set up EGM. Some of them will probably [...]

@robberthofmanfm Thanks. I have no issue installing EGM and getting it to work with  [this](https://github.com/ros-industrial/abb_robot_driver/tree/master/abb_robot_bringup_examples). Also, I could connect to the robot with KTH's repo, but because of the difference, I mentioned it is impossible to get it to work as I described. Do you have any Moveit samples with the  abb_robot_driver, please?

robberthofmanfm, 2022-06-28:
> @robberthofmanfm Thanks. I have no issue installing EGM and getting it to work with [this](https://github.com/ros-industrial/abb_robot_driver/tree/master/abb_robot_bringup_examples). Also, I could [...]

I'm afraid I'm not allowed to share that.