kth-ros-pkg/yumi #69: catkin_make error  in ros-melodic

ruoshiwen, 2021-09-29:
I want to use yumi robot with ros-melodic. I know this pkg is not suitable for melodic but I still try to bulid it. Everything goes well until I run "catkin_make -DCMAKE_BUILD_TYPE=Release" . I got following errors:

 /home/hit605/yumi_depends_ws/src/industrial_core/industrial_utils/src/param_utils.cpp:147:65: error: conversion from ‘urdf::JointLimitsSharedPtr {aka std::shared_ptr<urdf::JointLimits>}’ to non-scalar type ‘boost::shared_ptr<urdf::JointLimits>’ requested
     boost::shared_ptr<urdf::JointLimits> limits = iter->second->limits;

/home/hit605/yumi_depends_ws/src/yumi/yumi_hw/src/yumi_hw.cpp:185:79: error: conversion from ‘urdf::JointConstSharedPtr {aka std::shared_ptr<const urdf::Joint>}’ to non-scalar type ‘const boost::shared_ptr<const urdf::Joint>’ requested
   const boost::shared_ptr<const urdf::Joint> urdf_joint = urdf_model->getJoint(joint_name);
                                                           ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/yumi/yumi_hw/src/yumi_hw.cpp:186:89: error: conversion from ‘urdf::JointConstSharedPtr {aka std::shared_ptr<const urdf::Joint>}’ to non-scalar type ‘const boost::shared_ptr<const urdf::Joint>’ requested
   const boost::shared_ptr<const urdf::Joint> urdf_joint_sitffness = urdf_model->getJoint(joint_name + std::string("_stiffness"));
                                                                     ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/yumi/yumi_hw/src/yumi_hw.cpp:190:65: error: no matching function for call to ‘getJointLimits(const boost::shared_ptr<const urdf::Joint>&, joint_limits_interface::JointLimits&)’
    if (joint_limits_interface::getJointLimits(urdf_joint, limits))

/opt/ros/melodic/include/joint_limits_interface/joint_limits_urdf.h:48:13: note:   no known conversion for argument 1 from ‘const boost::shared_ptr<const urdf::Joint>’ to ‘urdf::JointConstSharedPtr {aka std::shared_ptr<const urdf::Joint>}’

/home/hit605/yumi_depends_ws/src/yumi/yumi_hw/src/yumi_hw.cpp:192:74: error: no matching function for call to ‘getSoftJointLimits(const boost::shared_ptr<const urdf::Joint>&, joint_limits_interface::SoftJointLimits&)’
    if (joint_limits_interface::getSoftJointLimits(urdf_joint, soft_limits))
                                                                          ^
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:48:18: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(this->updateConnection);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In member function ‘virtual void gazebo::MimicPlugin::UpdateChild()’:
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:88:42: error: ‘class gazebo::physics::Joint’ has no member named ‘GetAngle’; did you mean ‘GetForce’?
     mimic_joint_->SetPosition(0, joint_->GetAngle(0).Radian()*multiplier_);
                                          ^~~~~~~~
                                          GetForce
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I'm not sure whether the error  is the version of ros isn't match and I want to get some advices. Thank you very much.

diogoalmeida, 2021-09-29:
Hi,

Are you using the melodic branch?

ruoshiwen, 2021-09-29:
> Hi,
> 
> Are you using the melodic branch?

I am using the ros-melodic and I got the error when I run "catkin_make". I didn't recogenize

closed 2021-09-29 by ruoshiwen