kth-ros-pkg/yumi #71: cmake-error in melodic branch

jlbhb, 2021-09-30:
I tried to build this pkg on my yumi robot. when I cmake my pkg I met the error following:

`CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "optoforce_etherdaq_driver" with any of the following names:

    optoforce_etherdaq_driverConfig.cmake
    optoforce_etherdaq_driver-config.cmake

  Add the installation prefix of "optoforce_etherdaq_driver" to
  CMAKE_PREFIX_PATH or set "optoforce_etherdaq_driver_DIR" to a directory
  containing one of the above files.  If "optoforce_etherdaq_driver" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  yumi/yumi_optodaq/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hit605/yumi_depends_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hit605/yumi_depends_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed`

I think I have followed the wiki step by step, It looks like some files are needed, I would like to get some guide or advice to solve this error. Thanks a lot.

diogoalmeida, 2021-09-30:
Hi,

Try to see if installing https://github.com/OptoForce/etherdaq_ros solves your problem.

Best,
Diogo

jlbhb, 2021-09-30:
> https://github.com/OptoForce/etherdaq_ros

Thank you for your help, I followed your advice and the error disappeared. but I met a new error:

`/home/hit605/yumi_depends_ws/src/industrial_core/industrial_utils/src/utils.cpp:80:36: error: invalid initialization of reference of type ‘const joint_list& {aka const std::vector<boost::shared_ptr<urdf::Joint> >&}’ from expression of type ‘const std::vector<std::shared_ptr<urdf::Joint> >’
   const joint_list &joints = link->child_joints;
                              ~~~~~~^~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/industrial_core/industrial_utils/src/utils.cpp:101:34: error: invalid initialization of reference of type ‘const link_list& {aka const std::vector<boost::shared_ptr<urdf::Link> >&}’ from expression of type ‘const std::vector<std::shared_ptr<urdf::Link> >’
   const link_list &links = link->child_links;
                            ~~~~~~^~~~~~~~~~~
industrial_core/industrial_utils/CMakeFiles/industrial_utils.dir/build.make:62: recipe for target 'industrial_core/industrial_utils/CMakeFiles/industrial_utils.dir/src/utils.cpp.o' failed
make[2]: *** [industrial_core/industrial_utils/CMakeFiles/industrial_utils.dir/src/utils.cpp.o] Error 1
CMakeFiles/Makefile2:4674: recipe for target 'industrial_core/industrial_utils/CMakeFiles/industrial_utils.dir/all' failed
make[1]: *** [industrial_core/industrial_utils/CMakeFiles/industrial_utils.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[  3%] Linking CXX executable /home/hit605/yumi_depends_ws/devel/lib/optoforce_etherdaq_driver/etherdaq_subscriber
[  3%] Built target etherdaq_subscriber
[  3%] Linking CXX shared library /home/hit605/yumi_depends_ws/devel/lib/libetherdaq_driver.so
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In destructor ‘virtual gazebo::MimicPlugin::~MimicPlugin()’:
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:48:18: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(this->updateConnection);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In member function ‘virtual void gazebo::MimicPlugin::UpdateChild()’:
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:88:42: error: ‘class gazebo::physics::Joint’ has no member named ‘GetAngle’; did you mean ‘GetForce’?
     mimic_joint_->SetPosition(0, joint_->GetAngle(0).Radian()*multiplier_);
                                          ^~~~~~~~
                                          GetForce
[  3%] Built target etherdaq_driver
yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/build.make:62: recipe for target 'yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/src/gazebo_mimic_plugin/mimic_plugin.cpp.o' failed
make[2]: *** [yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/src/gazebo_mimic_plugin/mimic_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:4480: recipe for target 'yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/all' failed
make[1]: *** [yumi/gazebo_mimic/CMakeFiles/gazebo_mimic_plugin.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
I'm ensure that I have use the melodic branch and I have no idea how to deal with this. So greatful if more advices. Thank you very much.

diogoalmeida, 2021-09-30:
Just in case, if you checkout the master branch, do you get the same errors?

https://github.com/OptoForce/etherdaq_ros

jlbhb, 2021-09-30:
I tried to use master branch, I got another error:
`/home/hit605/yumi_depends_ws/src/industrial_core/industrial_utils/src/utils.cpp: In function ‘bool industrial_utils::findChainJointNames(const boost::shared_ptr<const urdf::Link>&, bool, std::vector<std::__cxx11::basic_string<char> >&)’:
/home/hit605/yumi_depends_ws/src/industrial_core/industrial_utils/src/utils.cpp:80:36: error: invalid initialization of reference of type ‘const joint_list& {aka const std::vector<boost::shared_ptr<urdf::Joint> >&}’ from expression of type ‘const std::vector<std::shared_ptr<urdf::Joint> >’
   const joint_list &joints = link->child_joints;
                              ~~~~~~^~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/industrial_core/industrial_utils/src/utils.cpp:101:34: error: invalid initialization of reference of type ‘const link_list& {aka const std::vector<boost::shared_ptr<urdf::Link> >&}’ from expression of type ‘const std::vector<std::shared_ptr<urdf::Link> >’
   const link_list &links = link->child_links;
                            ~~~~~~^~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In destructor ‘virtual gazebo::MimicPlugin::~MimicPlugin()’:
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:48:18: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(this->updateConnection);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp: In member function ‘virtual void gazebo::MimicPlugin::UpdateChild()’:
/home/hit605/yumi_depends_ws/src/yumi/gazebo_mimic/src/gazebo_mimic_plugin/mimic_plugin.cpp:88:42: error: ‘class gazebo::physics::Joint’ has no member named ‘GetAngle’; did you mean ‘GetForce’?
     mimic_joint_->SetPosition(0, joint_->GetAngle(0).Radian()*multiplier_);
                                          ^~~~~~~~
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
I think these two errors is samilar but different.

diogoalmeida, 2021-09-30:
Right, thanks for checking this out. It does seem like the melodic branch requires some improvements to work. You have two options: fix the melodic branch (we'd be thankful if you could make a PR with those changes) or, alternatively, try to use this package with ROS kinetic.

jlbhb, 2021-10-02:
> Right, thanks for checking this out. It does seem like the melodic branch requires some improvements to work. You have two options: fix the melodic branch (we'd be thankful if you could make a PR with those changes) or, alternatively, try to use this package with ROS kinetic.
> […](#)

Thank you for your help, maybe I will use ros kinetic.

jlbhb, 2021-10-02:
> Right, thanks for checking this out. It does seem like the melodic branch requires some improvements to work. You have two options: fix the melodic branch (we'd be thankful if you could make a PR with those changes) or, alternatively, try to use this package with ROS kinetic.
> […](#)

Hi  diogoalmeida ,I remember I followed  the tutorial on the wiki. There is a base pkg from  https://github.com/OrebroUniversity/industrial_core.git .And today I git clone it alone in another workspace. I got the same errors when I catkin_make. Do I need to change anything when I install the pkg if I want to use melodic branch?